Xilinx/Vitis-AI #724: the problem of YOLOv5 compilation

XiongDa0001, 2022-03-23:
When I use vai_c_tensorflow I encounter the following error：
[image]
The docker environment is Vitis-AI 2.0
vai_q_tensorflow can be executed successfully
I wonder how this problem should be solved?

zhenzhen-Xilinx, 2022-03-23:
Can you share your float model and quantized model so we can analyze the problem further. Thank you.

XiongDa0001, 2022-03-23:
> Thank you for your reply, my float model, frozen model and quantized model are all in my google drive.
https://drive.google.com/drive/folders/1PisOoT3rBbX2XAJVwosGW0akKk87oLrx?usp=sharing

zhenzhen-Xilinx, 2022-03-25:
Hi @XiongDa0001， Based on the model you provided, I reproduced the problem below. The solution is that the shape of the input can be less than 1.

[image]

Because the problem I reproduced is not the same as what you asked. Can you offered the float and quantized model again. Thanks.

XiongDa0001, 2022-03-25:
> Hi @XiongDa0001， Based on the model you provided, I reproduced the problem below. The solution is that the shape of the input can be less than 1.
> 
> [image]
> 
> Because the problem I reproduced is not the same as what you asked. Can you offered the float and quantized model again. Thanks.

My float and quantized model in google dive.
https://drive.google.com/drive/folders/1PisOoT3rBbX2XAJVwosGW0akKk87oLrx?usp=sharing

XiongDa0001, 2022-03-25:
> Hi @XiongDa0001， Based on the model you provided, I reproduced the problem below. The solution is that the shape of the input can be less than 1.
> 
> [image]
> 
> Because the problem I reproduced is not the same as what you asked. Can you offered the float and quantized model again. Thanks.

The problem you have encounterd was not define the command{--options '{"input_shape": "1,416,416,3"}'}
[image]

zhenzhen-Xilinx, 2022-04-02:
hi @XiongDa0001 ,Sorry for late reply. The model contains tf.strided_slice. Now we only supports the case where the tf.strided_slices parameter ellipsis_mask = 0. However in this model the parameter ellipsis_mask = 1. tf.strided_slice belongs to the preprocessing part of the model. Could you removes the preprocessing
?
[image]

XiongDa0001, 2022-04-02:
> hi @XiongDa0001 ,Sorry for late reply. The model contains tf.strided_slice. Now we only supports the case where the tf.strided_slices parameter ellipsis_mask = 0. However in this model the parameter [...]

Thank you very much, I'll remove this it.

closed 2022-04-02 by XiongDa0001

zhenzhen-Xilinx, 2022-04-02:
hi @XiongDa0001 , If you still have any questions, please feel free to reopen it. Thank you very much.