Xilinx/Vitis-AI #725: Error when compiling quantized pytorch model with vai_c_xir: [UNILOG][FATAL][XCOM_DATA_OUTRANGE][Data value is out of range!] The dynamic space is too small for your data size. Please recheck the bankgroup assigner or tiling process.

MarvS95, 2022-03-23:
Hello,

I quantized a pytorch model using a calibration dataset of 100 training video clips and got an .xmodel. During compilation, the 27 GB of RAM is completely exhausted and after 15 minutes I get the following error:

`[UNILOG][FATAL][XCOM_DATA_OUTRANGE][Data value is out of range!] The dynamic space is too small for your data size. Please recheck the bankgroup assigner or tiling process.`

 
Has anyone had the same error or knows what might be causing this error message? The size of the quantized .xmodel is 136,6 MB. For comparison I compiled ResNet50_int.xmodel which has a size of 102 MB. Compiling took only a few seconds.

Hope you can help me.
Thank you!

wanghong4compiler, 2022-03-23:
Hi @MarvS95 ,
which version of vitis-ai did you use? and could you share the xmodel used for compilation? so that I can reproduce this issue, thank you.

MarvS95, 2022-03-23:
Hi @wanghong4compiler,

Thank you for your help.
I use Vitis-AI Docker 2.0.0.1103. How can I share the xmodel? It's too big to share it here.

wanghong4compiler, 2022-03-23:
Hi @MarvS95 ，
Is it easy to upload to onedrive or network disk? 
Just let me download the model.
thank you.

MarvS95, 2022-03-23:
Hi @wanghong4compiler,

here is the quantized xmodel:

https://1drv.ms/u/s!AlahwTvPqgAlggH4WB9Di2OzLSZ1?e=Pr1nlz

wanghong4compiler, 2022-03-25:
Hi @MarvS95 ，
Thank you for sharing the xmodel.
There is one more question, which type of arch did you use?

MarvS95, 2022-03-25:
Hi @wanghong4compiler ,

I use DPUCZDX8G/ZCU102

Thank you

MarvS95, 2022-03-25:
B4096

MarvS95, 2022-03-29:
Hi @wanghong4compiler,

were you able to compile the xmodel?

wanghong4compiler, 2022-03-30:
Hi @MarvS95 ,
This issue has been reproduced. It seems like this model is a 3D model, currently, only DPUCVDX8G can support 3D ops.

closed 2022-06-11 by qianglin-xlnx