Xilinx/Vitis-AI #726: [UNILOG][FATAL][XCOM_ACGEN_ERROR][instuction generating fail, please contact us.]

JaviMota, 2022-03-23:
Hi 

Im trying to compile an own model following the 05-Keras_FCN8_UNET_segmentation tutorial using the FCN8 network and I get this error:
[UNILOG][FATAL][XCOM_ACGEN_ERROR][instuction generating fail, please contact us.] 
I only changed the dataset by an own one and the number of classes to 2. The neural network is the same you propose. The weird thing is it has compiled one of several attempts without changing anything. Im using Vitis-ai 2.0 in vitis-ai-tensorflow environment.
I have compiled the tutorial with the original classes and it works.
Do you know what can be the problem?

Thank you.

wanghong4compiler, 2022-03-24:
Hi @JaviMota ,
which version and which hardware architecture did you use?
And, could you share the xmodel used for compiler?
So that I can reproduce this issue.
Thank you.

JaviMota, 2022-03-24:
Hi
Vitis-AI Docker version: 2.0.0.1103
Architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
Environment: conda activate vitis-ai-tensorflow

I share with you the quantized model as Im not able to get the xmodel because of the error.
https://drive.google.com/file/d/1wIl82XzjjTUxkMAMmm37vjdLHnLJKclA/view?usp=sharing

Thank you

JaviMota, 2022-03-28:
Hi 
To add a  bit more info I have tried to run the 05-Keras_FCN8_UNET_segmentation tutorial with several number of classes and this is what I got:

2,3 classes --> Error in the compilation
4 classes --> After a while, error in the compilation
5 to 12 classes --> It works fine

Could be a bug or Am I doing something wrong?

Thank you.

JaviMota, 2022-04-04:
Hi
Any news on this topic? have you tried to compile the model?
Thank you.

wanghong4compiler, 2022-07-13:
Hi @ZiYinHong ,
This issue has been fixed in VAI2.5, could you update the toolchain and try this again?

JaviMota, 2022-07-14:
Hi

I have tested it with the latest version of Vitis-AI 2.5 and Im still having the issue.
I used to test the https://github.com/Xilinx/Vitis-AI-Tutorials/tree/1.4/Design_Tutorials/05-Keras_FCN8_UNET_segmentation tutorial and I only changed the fcn_config.py to this:

###############################################################################
# global variables
###############################################################################

#Size of images
WIDTH  = 224 
HEIGHT = 224 

#normalization factor
NORM_FACTOR = 127.5

#number of classes
NUM_CLASSES = 2

# names of classes
CLASS_NAMES = ("Sky",
               "Wall")

BATCH_SIZE = 32
EPOCHS = 5

Im also using the DPUCZDX8G_ISA1_B512 architecture.
The file Im running is run_fcn8.sh

Regards.