Xilinx/Vitis-AI #727: Config PMIC fails

ash8327, 2022-03-23:
I have built and ran the tutorial with no modification successfully but when using Petalinux to configure the Common Software the dpu_sw_optimize.sh fails with:

Auto resize ext4 partition ...[✔]
Start QoS config ...[✔]
Config PMIC irps5401 .../dev/i2c-4: Device or resource busy
Couldn't attach to device 0x43

I included the i2ctool in my Petalinux build but this still errors out. I am using the same base platform as the tutorial.

tianfang-fafafa, 2022-03-24:
hi @ash8327  which version and tutorial are you using?

ash8327, 2022-03-24:
Hi @tianfang-fafafa , I have tried both 2021.2 and 2021.1. I noticed there used to be petalinux update where dpu_sw_optimize.sh is and I also rebuilt my petalinux image with those changes but I still get the same error.
Is that update applicable to the new tutorial?

tianfang-fafafa, 2022-03-25:
@ash8327 ，could you please provide tutorial link and your step to reproduce?

qianglin-xlnx, 2022-03-25:
HI @ash8327 
Do you use ZCU104 board? irps5401 is the power patch and can only be used for ZCU104 board.

qianglin-xlnx, 2022-06-11:
Hi @ash8327 
Has the problem been solved?

closed 2022-06-15 by qianglin-xlnx