Xilinx/Vitis-AI #729: dpu_trd buid failure

flynn27, 2022-03-25:
i meet some problem when i build the official project,i get the bsp file from this link:https://github.com/Xilinx/Vitis-AI/blob/1.3.1/dsa/DPU-TRD/prj/Vivado
and the tools i used is vivado 2020.2,petalinux 2020.2,i use the command petalinux build and failed,the problems and build.log as followed.
i followed the readme but i do not konw why i can not pass it.Thank you for you help.

ERROR: dpu-1.0-r0 do_compile: oe_runmake failed
ERROR: dpu-1.0-r0 do_compile: Execution of '/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/temp/run.do_compile.32260' failed with exit code 1:
make -C /home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-source M=/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0
make[1]: Entering directory '/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-source'
make[2]: Entering directory '/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-build-artifacts'
  CC [M]  /home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/dpu.o
/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/dpu.c: In function 'xlnx_dpu_alloc_bo':
/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/dpu.c:509:6: error: implicit declaration of function 'get_user'; did you mean 'get_super'? [-Werror=implicit-function-declaration]
  509 |  if (get_user(size, &req->size)) {
      |      ^~~~~~~~
      |      get_super
/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/dpu.c:516:6: error: implicit declaration of function 'put_user' [-Werror=implicit-function-declaration]
  516 |  if (put_user(pb->capacity, &req->capacity)) {
      |      ^~~~~~~~
/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/dpu.c: In function 'xlnx_dpu_ioctl':
/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/dpu.c:656:7: error: implicit declaration of function 'copy_from_user'; did you mean 'sg_copy_from_buffer'? [-Werror=implicit-function-declaration]
  656 |   if (copy_from_user(&t, (void *)arg,
      |       ^~~~~~~~~~~~~~
      |       sg_copy_from_buffer
/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/dpu.c:677:7: error: implicit declaration of function 'copy_to_user' [-Werror=implicit-function-declaration]
  677 |   if (copy_to_user(data, &t, sizeof(struct ioc_kernel_run_t)))
      |       ^~~~~~~~~~~~
/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/dpu.c: In function 'xlnx_dpu_probe':
/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/dpu.c:971:15: error: implicit declaration of function 'devm_platform_get_and_ioremap_resource'; did you mean 'devm_platform_ioremap_resource'? [-Werror=implicit-function-declaration]
  971 |  xdpu->regs = devm_platform_get_and_ioremap_resource(pdev, 0, &res);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |               devm_platform_ioremap_resource
/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/dpu.c:971:13: warning: assignment to 'void *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  971 |  xdpu->regs = devm_platform_get_and_ioremap_resource(pdev, 0, &res);
      |             ^
cc1: some warnings being treated as errors
make[3]: *** [/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-source/scripts/Makefile.build:266: /home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/dpu.o] Error 1
make[2]: *** [/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-source/Makefile:1652: /home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0] Error 2
make[2]: Leaving directory '/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-build-artifacts'
make[1]: *** [Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-source'
make: *** [Makefile:9: all] Error 2
WARNING: /home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/temp/run.do_compile.32260:1 exit 1 from 'exit 1'

ERROR: Logfile of failure stored in: /home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/temp/log.do_compile.32260
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8 KERNEL_SRC=/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-source KERNEL_PATH=/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-source KERNEL_VERSION=5.4.0-xilinx-v2020.2 CC=aarch64-xilinx-linux-gcc   -fuse-ld=bfd -fmacro-prefix-map=/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0=/usr/src/debug/dpu/1.0-r0                      -fdebug-prefix-map=/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0=/usr/src/debug/dpu/1.0-r0                      -fdebug-prefix-map=/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/recipe-sysroot-native=  -fdebug-prefix-map=/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-source=/usr/src/kernel LD=aarch64-xilinx-linux-ld.bfd   AR=aarch64-xilinx-linux-ar  O=/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-build-artifacts KBUILD_EXTRA_SYMBOLS=
| make -C /home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-source M=/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0
| make[1]: Entering directory '/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-source'
[... cut ...]
| make[1]: Leaving directory '/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work-shared/zynqmp-generic/kernel-source'
| make: *** [Makefile:9: all] Error 2
| WARNING: /home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/build/tmp/work/zynqmp_generic-xilinx-linux/dpu/1.0-r0/temp/run.do_compile.32260:1 exit 1 from 'exit 1'
| 
ERROR: Task (/home/lab601/xilinx/dpu/Vitis-AI-1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/xilinx-zcu102-trd/project-spec/meta-user/recipes-modules/dpu/dpu.bb:do_compile) failed with exit code '1'

andrew0917, 2022-04-11:
Hi @flynn27,
     I download the bsp from [https://github.com/Xilinx/Vitis-AI/blob/1.3.1/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/download_bsp.sh](url), and found that the bsp is the latest release vai2.0. It maybe overwrited by the new release bsp version when use the same download link, I will ask for my colleagues help me update the right download link with specific verison number.
     In order to fix your issue quicker, this is the vai1.3.1 release bsp [xilinx-zcu102-trd.zip](https://github.com/Xilinx/Vitis-AI/files/8462706/xilinx-zcu102-trd.zip), you can use it to build the petalinux project in vai1.3.1.

flynn27, 2022-04-11:
> 

thank you for your help,as i can not deal with this problem,i use vivado 2021.1,petalinux2021.1 and vai1.4,i found this can build pass,but i can not run resnet50 demo,and has such problem,whether you can locate the problem from.
the problem is dpu_controller_dnndk.cpp, run dpu failed

andrew0917, 2022-04-13:
Hi @flynn27,
Could you please run the following command to check if DPU is integrated properly?

root@xilinx-zcu102-2021_1:# lsmod
Module Size Used by
dmaproxy 16384 0
dpcma 16384 0
mali 270336 0
dpu 40960 0
root@xilinx-zcu102-2021_1:# cat /proc/interrupts | grep dpu
57: 1 0 0 0 GICv2 138 Edge dpu_isr
58: 0 0 0 0 GICv2 139 Edge dpu_isr
59: 0 0 0 0 GICv2 140 Edge dpu_isr
65: 0 0 0 0 GICv2 142 Edge dpu_smfc
root@xilinx-zcu102-2021_1:~# show_dpu
device_core_id=0 device= 0 core = 0 fingerprint = 0x1000020f6014407 batch = 1 full_cu_name=unknown:dpu0

root@xilinx-zcu102-2021_1:~# env DEBUG_DPU_CONTROLLER=1 show_dpu
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1022 07:56:24.582245 11357 dpu_controller.cpp:38] add factory method 00_dnndk
I1022 07:56:24.582329 11357 dpu_controller_dnndk.cpp:192] register the dnndk dpu controller
I1022 07:56:24.582594 11357 dpu_control_xrt.cpp:60] no xrt device found, xrt no dpu controller registered.
I1022 07:56:24.582690 11357 dpu_controller_dnndk.cpp:60] data 0xffffeaa98990 offset 2399142384 size 4
I1022 07:56:24.582739 11357 dpu_controller_dnndk.cpp:60] data 0xffffeaa989a8 offset 2399142388 size 4
I1022 07:56:24.582769 11357 dpu_controller_dnndk.cpp:96] fingerprint: 0x1000020f6014407 0x1000020 0xf6014407
I1022 07:56:24.582792 11357 dpu_controller.cpp:49] create dpu controller via 00_dnndk ret= 0xaaaadc026750
device_core_id=0 device= 0 core = 0 fingerprint = 0x1000020f6014407 batch = 1 full_cu_name=unknown:dpu0

root@xilinx-zcu102-2021_1:~# xdputil query
{
"DPU IP Spec":{},
"VAI Version":{
"libvart-runner.so":"Xilinx vart-runner Version: 1.4.0-5d7e19a2a47a01369a0d29e8f28996035704de6c 117 2021-06-29-15:02:25 ",
"libvitis_ai_library-dpu_task.so":"Xilinx vitis_ai_library dpu_task Version: 1.4.0-401ecb06da9ad26383f47b531c911638d4a0745d 117 2021-06-29 07:07:57 [UTC] ",
"libxir.so":"Xilinx xir Version: xir-1effe8a9815a8f3b66c4739d712573abff59c63d 2021-06-29-14:58:38",
"target_factory":"target-factory.1.4.0 ce1b39e329cc06cb7545e8aa39174fb8b9969f0b"
},
"kernels":[
{
"DPU Arch":"DPUCVDX8G_ISA0_B4096_01000050F6014407",
"DPU Frequency (MHz)":325,
"cu_addr":"0x8f000000",
"cu_idx":0,
"fingerprint":"0x1000050f6014407",
"is_vivado_flow":true,
"name":"DPU Core 0"
}
]
}

flynn27, 2022-04-14:
> 
Hi,thank you for your reply,this is my dpu information depends on your suggestions.After this,my problem message is also showed below.hopes for your help!
dpu messgae
root@xilinx-zcu104-2021_1:~# cat /proc/interrupts |grep dpu
 56:          0          0          0          0     GICv2 138 Level     dpu_isr
root@xilinx-zcu104-2021_1:~# lsmod
Module                  Size  Used by
dmaproxy               16384  0
mali                  270336  0
dpcma                  16384  0
dpu                    40960  0
uio_pdrv_genirq        16384  0
root@xilinx-zcu104-2021_1:~# show_dpu
device_core_id=0 device= 0 core = 0 fingerprint = 0x1000020f6014407 batch = 1 full_cu_name=unknown:dpu0

root@xilinx-zcu104-2021_1:~# env DEBUG_DPU_CONTROLLER=1 show_dpu
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0414 07:22:08.241757  1326 dpu_controller.cpp:38] add factory method 00_dnndk
I0414 07:22:08.241866  1326 dpu_controller_dnndk.cpp:192] register the dnndk dpu controller
I0414 07:22:08.242164  1326 dpu_control_xrt.cpp:60] no xrt device found, xrt no dpu controller registered.
I0414 07:22:08.242249  1326 dpu_controller_dnndk.cpp:60] data 0xffffcdec8508 offset 2399142384 size 4
I0414 07:22:08.242297  1326 dpu_controller_dnndk.cpp:60] data 0xffffcdec8510 offset 2399142388 size 4
I0414 07:22:08.242329  1326 dpu_controller_dnndk.cpp:96]  fingerprint: 0x1000020f6014407 0x1000020 0xf6014407
I0414 07:22:08.242355  1326 dpu_controller.cpp:49] create dpu controller via 00_dnndk ret= 0xaaaacfb317c0
device_core_id=0 device= 0 core = 0 fingerprint = 0x1000020f6014407 batch = 1 full_cu_name=unknown:dpu0

root@xilinx-zcu104-2021_1:~# xdputil query
{
    "DPU IP Spec":{},
    "VAI Version":{
        "libvart-runner.so":"Xilinx vart-runner Version: 1.4.0-fa49b842f283242091476cf8e1ae4d242a2a838e 585 2021-07-13-18:50:52 ",
        "libvitis_ai_library-dpu_task.so":"Xilinx vitis_ai_library dpu_task Version: 1.4.0-93d2e0097889ccebb5f1256e0afe8409de81f482 585 2021-07-13 10:54:30 [UTC] ",
        "libxir.so":"Xilinx xir Version: xir-ff89b11dcabb00eef6d148fcf660c8e6d02eb184 2021-07-13-18:49:08",
        "target_factory":"target-factory.1.4.0 ce1b39e329cc06cb7545e8aa39174fb8b9969f0b"
    },
    "kernels":[
        {
            "DPU Arch":"DPUCVDX8G_ISA0_B4096_01000050F6014407",
            "DPU Frequency (MHz)":325,
            "cu_addr":"0x8f000000",
            "cu_idx":0,
            "fingerprint":"0x1000050f6014407",
            "is_vivado_flow":true,
            "name":"DPU Core 0"
        }
    ]
}
error information
dpu_controller_dnndk.cpp:101] Check failed: retval == 0 (-1 vs. 0) run dpu failed.
[15:21:36:490] *** Check failure stack trace: ***

zz002, 2022-04-18:
@flynn27 The problem of wrong xdputil query fingerprint data has been resolved and will be updated in vai2.5 version.

qianglin-xlnx, 2022-04-18:
Hi @flynn27 
I think you can skip the "xdputil query" error before we fix it and directly run the VART samples. Could you please try this? Thank you.

flynn27, 2022-04-19:
> 

hi,zz002,thank you for your reply,this is means that i can not use VAI1.4 and petalinux 2021.1 to creat dpu images in zcu104?

andrew0917, 2022-04-21:
Hi @flynn27,
    Could your run dmesg command after runing resnet50 demo to show the driver log, BTW, do your customize your DPU IP design, such as mofify the DPU base address.

qianglin-xlnx, 2022-06-12:
Hi @andrew0917 
Has the issue been solved?

closed 2022-06-23 by qianglin-xlnx