Xilinx/Vitis-AI #730: Versal DPU on DFX platform - kernel panic

ksstms, 2022-03-26:
I'm trying to run the Versal DPU on the official VCK190 DFX platform. Actually I'm using an ES board, but I got both designs working great separately.

I modified the XVDPU-TRD's Vitis project a little bit, because the SP-tags of the DFX platform differ. It builds fine, and the generated Vivado project looks fine.

I have issues on the software side though. First I tried to add `xrt` and `packagegroup-petalinux-vitisai` to the DFX platform's PetaLinux project. That didn't work unfortunately:

```
root@xilinx-vck190-2021_2:~# xdputil benchmark resnet50-32-1.xmodel -i -1 2 
Traceback (most recent call last): 
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main 
    return _run_code(code, main_globals, None, 
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code 
    exec(code, run_globals) 
  File "/usr/lib/python3.8/site-packages/xdputil.py", line 44, in <module> 
    main()
  File "/usr/lib/python3.8/site-packages/xdputil.py", line 38, in main 
    args.func(args) 
  File "/usr/lib/python3.8/site-packages/xdputil_component/benchmark.py", line 45, in main 
    graph = xir.Graph.deserialize(args.xmodel) 
RuntimeError: bad any_cast 
```

This is similar to issue #675, so I guessed there is something wrong with the VART version. I noticed that it isn't just added as a packagegroup in the XVDPU-TRD, but it has its [own recipes](https://github.com/Xilinx/Vitis-AI/tree/master/dsa/XVDPU-TRD/vck190_platform/petalinux/xilinx-vck190-base-trd/project-spec/meta-user/recipes-vitis-ai).  Then I tried two things: 
 1. add all of those to the DFX  platform's PetaLinux project
 2. modify the XVDPU-TRD project to use the DFX project's static xsa and device tree files. 

Both approaches resulted in the same kernel panic on boot:

<details>
<summary>Spoiler</summary>

```
Starting rpcbind daemon...done.
starting statd: done
Starting internet superserver: inetd.
NFS daemon support not enabled in kernel
Starting ntpd: done
Starting syslogd/klogd: done
Starting internet superserver: xinetd.
daemon not start due to lack of /dev/watchdog
[   22.506341] SError Interrupt on CPU0, code 0xbf000000 -- SError
[   22.506343] CPU: 0 PID: 777 Comm: devmem Tainted: G           O      5.10.0-xilinx-v2021.2 #1
[   22.506345] Hardware name: Xilinx Versal vck190 Eval board revA (QSPI) (DT)
[   22.506346] pstate: 60000000 (nZCv daif -PAN -UAO -TCO BTYPE=--)
[   22.506348] pc : 000000000049bff0
[   22.506349] lr : 000000000040f604
[   22.506350] sp : 0000ffffe85e7d50
[   22.506351] x29: 0000ffffe85e7d50 x28: 0000000000000000 
[   22.506355] x27: 00000000005a5000 x26: 0000000000000001 
[   22.506358] x25: 00000000005a0260 x24: 0000000000000000 
[   22.506361] x23: 00000000005a3728 x22: 000000000059f000 
[   22.506364] x21: 00000000005a3000 x20: 0000ffffe85e8f09 
[   22.506367] x19: 00000000005a3728 x18: 0000000000000000 
[   22.506370] x17: 0000000000000000 x16: 0000000000000000 
[   22.506373] x15: 00000000000000b0 x14: 0000000000000000 
[   22.506375] x13: 0000ffffa6bfb150 x12: 64762d78756e696c 
[   22.506378] x11: ffffffffffffffff x10: 0000000000000008 
[   22.506381] x9 : 0000000000000007 x8 : 00000000000000de 
[   22.506384] x7 : 0000000000000020 x6 : 0000000000002000 
[   22.506387] x5 : 0000020000071000 x4 : 0000000000000000 
[   22.506390] x3 : 0000000000000001 x2 : 00000000005a3728 
[   22.506393] x1 : 0000000000000001 x0 : 0000000000000000 
[   22.506396] Kernel panic - not syncing: Asynchronous SError Interrupt
[   22.506398] CPU: 0 PID: 777 Comm: devmem Tainted: G           O      5.10.0-xilinx-v2021.2 #1
[   22.506399] Hardware name: Xilinx Versal vck190 Eval board revA (QSPI) (DT)
[   22.506400] Call trace:
[   22.506402]  dump_backtrace+0x0/0x190
[   22.506403]  show_stack+0x18/0x30
[   22.506404]  dump_stack+0xd4/0x110
[   22.506405]  panic+0x15c/0x324
[   22.506407]  add_taint+0x0/0xb0
[   22.506408]  arm64_serror_panic+0x78/0x84
[   22.506409]  do_serror+0x3c/0x6c
[   22.506410]  el0_error_naked+0x14/0x1c
[   22.506425] SMP: stopping secondary CPUs
[   22.506426] Kernel Offset: disabled
[   22.506427] CPU features: 0x0040022,21002008
[   22.506429] Memory Limit: none
```

</details>

I'm not really sure what to check at this point, so I would be very grateful for any ideas you have.

qianglin-xlnx, 2022-03-28:
Hi @ksstms 
For VART2.0, you can download the prebuilt packages and install them. Find them in: 
https://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc/VART/2021.2

ksstms, 2022-03-31:
I noticed that the official DFX platform's interrupt controller is configured with edge-sensitive inputs. Since the DPU has a level interrupt (#608), I changed those in the platform. (It's interesting though, that this didn't bother the `vadd` and `aie_adder` kernels. I guess those examples produce only a single interrupt, and the edge gets registered correctly.)

Now the system doesn't freeze after using xbutil + workaround script + benchmark, but it produces the same timeout that I mentioned previously:

```
root@xilinx-vck190-2021_2:~# xdputil benchmark resnet50-32-1.xmodel -i -1 2
XAIEFAL: INFO: Resource group Avail is created.
XAIEFAL: INFO: Resource group Static is created.
XAIEFAL: INFO: Resource group Generic is created.
WARNING: Logging before InitGoogleLogging() is written to STDERR
W0330 07:20:14.015880  1104 xrt_cu.cpp:188] cu timeout! device_core_idx 0  handle=0xffffacd456c0 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10009  bo=1 is_done 0 
I0330 07:20:14.015941  1104 xrt_cu.cpp:99] Total: 10009638us	ToDriver: 18446743934700048us	ToCU: 0us	Complete: 0us	Done: 149019141us
F0330 07:20:14.015962  1104 dpu_control_xrt_xv_dpu.cpp:193] dpu timeout! core_idx = 0
 AP 1  LSTART 20  LEND 19  CSTART 1  CEND 1  SSTART 0  SEND 0  MSTART 1  MEND 1  CYCLE_L 1000973087  CYCLE_H 0 
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20:  1101 Aborted                 /usr/bin/python3 -m xdputil $*
```

JasonXijieJia, 2022-04-01:
Hi @ksstms 
  Your log indicates that the xvdpu is hanging while buffering data from ddr to on-chip ram via noc.
  Would you please check if there is anything different about you noc settings in the dfx platform and the trd platform.
  Thank you for your feedback.

ksstms, 2022-04-08:
Hi @JasonXijieJia 
Thanks for the tip! Sorry for the late reply, I had some issues getting an ILA to work. 
So, I checked the AXI interfaces, and the `WGT` ones seem to have a problem with reading.
 - The AR channel is okay
 - `RREADY` goes high after the address is transmitted
 - `RVALID` never arrives (tried to trigger on that, but nothing came)

[image]

This is the saved ILA data in case you want to check it out: [xvdpu_ila.zip](https://github.com/Xilinx/Vitis-AI/files/8451861/xvdpu_ila.zip)

The `WGT` interfaces (and all others for that matter) are reading from `0x0006E0F4000` and similar addresses, which are valid DDR addresses. I checked the address editor in the Vivado project, and all interfaces have correctly assigned DDR segments.

The following table shows the NoC connectivity. The `WGT` connections are spread across the 4 memory controller ports, and they are also shared with the working interfaces, so I don't know why they don't work.

|   | M00_INI / MC0 | M01_INI / MC1 | M02_INI / MC2 | M03_INI / MC3 | Read BW | Write BW |
|---|:-:|:-:|:-:|:-:|:-:|:-:|
| M00_INSTR | X |   |   |   | 64  | 32 |
| M00_BIAS  |   | X |   |   | 1000| 32 |
| M00_WGT   |   |   | X |   | 6000 | 32 |
| M01_WGT   |   |   |   | X | 6000 | 32 |
| M02_WGT   | X |   |   |   | 6000 | 32 |
| M03_WGT   |   | X |   |   | 6000 | 32 |
| M00_IMG   |   |   | X |   | 1000 | 32 |
| M01_IMG   |   |   |   | X | 1000 | 32 |

ksstms, 2022-04-12:
I poked around a bit more, and found that the `WGT` ports have their NoC QoS read traffic class set to `LOW_LATENCY` whereas the others are left at the default `BEST_EFFORT`. 

I removed the traffic class setting of these ports from the [post link script](https://github.com/Xilinx/Vitis-AI/blob/0c57c1fe93ac35967ec4c273aa38a9d0a8bdc006/dsa/XVDPU-TRD/vitis_prj/scripts/post_linker.tcl#L36) and the NoC ports started working. The benchmark runs fine now, but unfortunately this change causes a 25% FPS decrease on ResNet50 using the C32B1 (smallest) DPU configuration. 

The QoS setting is vital to the DPU performance. I also tested the original XVDPU-TRD (no DFX) with the `BEST_EFFORT` QoS setting, and it also resulted in an FPS decrease.

Do you have any tips on getting this working on the DFX platform?

xlinyg, 2022-04-13:
Hi, @ksstms 
More better DDR reading bandwith for DPU's WGT ports, more better performance of DPU.
Traffci class 'LOW_LATENCY' will have better DDR reading performance than ' BEST_EFFORT'. We have same test result as yours when changing WGT as ' BEST_EFFORT'.

For DFX platform, to makesure its static region has higher prority reading DDR than kernel in dynamic region, maybe not allow to set reading traffic class as 'LOW_LATENCY' for kernel in dynamic region.

ksstms, 2022-04-13:
Hi @xlinyg 
Yes, I suspect the same. I also asked this question in a [support forum thread](https://support.xilinx.com/s/question/0D52E000077QtmpSAC/changing-versal-noc-qos-in-dfx-design-breaks-nmu?language=en_US). I was hoping that someone here has a solution for it, as it is very rare to get any useful answers on the forum.

@qianglin-xlnx Thanks again for the advice about installing the prebuilt packages. That solved the software-related problems. I'm still interested if I can add those prebuilt packages or build the same specific version in my PetaLinux project. Thanks for helping.

qianglin-xlnx, 2022-04-19:
Hi @ksstms 
You can refer to https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Recipes to add the VART in your PetaLinux project. You can also modify the source code and build the specific version.

ksstms, 2022-04-22:
Thanks @qianglin-xlnx! That works well.

closed 2022-06-15 by qianglin-xlnx