Xilinx/Vitis-AI #731: Ubuntu 18: Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103

krishnaadithya, 2022-03-27:
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/g/glibc/libc6_2.27-3ubuntu1.5_amd64.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA512:84869a5cf0903b1045c7e7e58c6bfcf9800c40fdc8b651087fa448ea2af1c18158df29b92af6dd0113ea77e627a23d7a4da4bae683db6d33f8d9104f044f207f
    - SHA256:c4af07dc8a2fdc9c4f25f103cd03bcca7231c19d9ac9171fa7eb5eecb7fc46d7
    - SHA1:47e5405d0f9c7ff40daff7ec010ff816d901f35e [weak]
    - MD5Sum:aef43cd3a8359820eb9a027c19a056c8 [weak]
    - Filesize:2829792 [weak]
   Hashes of received file:
    - SHA512:1fddf82ff4934c44911550bfd1957def5e01a22d58e867733c29437fe59d46b71a14c672f505d1ae97a2630cf35e983964543397ce05c52f31670b7bcbe10531
    - SHA256:cbc0dcc4803d2979d9b5e734ae2a4d45e20a1aaec8cfc895a2209285a9ff7573
    - SHA1:c7856f53283b18ae2a3e4d387d491752fbb4a6ec [weak]
    - MD5Sum:2c6e3e6f7742330dc915c33ed3dbbd51 [weak]
    - Filesize:2829792 [weak]
   Last modification reported: Tue, 01 Mar 2022 15:23:49 +0000
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/b/boost1.65.1/libboost1.65-tools-dev_1.65.1+dfsg-0ubuntu5_amd64.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:157a8911db429eb31e30da259343c4f1a2f727e17471a75fb82137583928cf0c
    - SHA1:a0bfe685224854fdf4a2e9b6d5c280afac928634 [weak]
    - MD5Sum:d94934e833617039c0b2c99d992cf35b [weak]
    - Filesize:1221844 [weak]
   Hashes of received file:
    - SHA256:3f85b524dfef9e29de52e9e63b162f02673607f13f3cbe7c2d3a0537d383485e
    - SHA1:fb91f99a8d8fe061c5bc7d6914b01656c6ddf166 [weak]
    - MD5Sum:38c0cb861d9e21fffe4429a97aacd16d [weak]
    - Filesize:1221844 [weak]
   Last modification reported: Tue, 06 Mar 2018 11:39:14 +0000
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/l/lapack/liblapack3_3.7.1-4ubuntu1_amd64.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:237d103dfb157e5e056d4442bca48fa20065f448c1626091245f247e4ecf3a9e
    - SHA1:4ec796932fb33066df188feac021af8f5733f3bb [weak]
    - MD5Sum:e11187e0bcd7ee0a5dd04416828693ff [weak]
    - Filesize:2112388 [weak]
   Hashes of received file:
    - SHA256:6934a0719c4949a28e0f282265907f9e6cdfb30889aff7a188f48fb73241ef9e
    - SHA1:8986b7c050a1f02cf3239660f19d150601db2c33 [weak]
    - MD5Sum:eef44c1ad948b3aeb0d375394a43db53 [weak]
    - Filesize:2112388 [weak]
   Last modification reported: Tue, 28 Nov 2017 20:19:03 +0000
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
The command '/bin/bash -c chmod 1777 /tmp     && mkdir /scratch     && chmod 1777 /scratch     && apt-get update -y     && apt-get install -y --no-install-recommends         apt-transport-https         autoconf         automake         bc         build-essential         bzip2         ca-certificates         curl         g++         gdb         git         gnupg         libboost-all-dev         libgflags-dev         libgoogle-glog-dev         libgtest-dev         libjson-c-dev         libjsoncpp-dev         libssl-dev         libtool         libunwind-dev         locales         make         openssh-client         openssl         python3         python3-dev         python3-minimal         python3-numpy         python3-pip         python3-setuptools         python3-venv         software-properties-common         sudo         tree         unzip         vim         wget         yasm         zstd' returned a non-zero code: 100
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103
Error response from daemon: No such image: xilinx/vitis-ai-gpu:2.0.0.1103

hanxue, 2022-03-28:
Hi @krishnaadithya 

There seems to be network error when downloading Ubuntu packages. We have seen this happening before, and usually a docker rebuild will solve the problem. 

If this error persist, can you please try to download the Ubuntu package on the host you are running `docker build` and verify the checksum?

For example

```
wget http://archive.ubuntu.com/ubuntu/pool/main/g/glibc/libc6_2.27-3ubuntu1.5_amd64.deb
sha256sum libc6_2.27-3ubuntu1.5_amd64.deb
```

krishnaadithya, 2022-03-28:
@hanxue  sorry I have not used dockers much, how do I rebuild a docker? 

and I tried doing these two steps:
wget http://archive.ubuntu.com/ubuntu/pool/main/g/glibc/libc6_2.27-3ubuntu1.5_amd64.deb
sha256sum libc6_2.27-3ubuntu1.5_amd64.deb

But I get the same error

hanxue, 2022-03-28:
Hi @krishnaadithya ,

I meant re-run the `docker_build_gpu.sh` script. 

What is the output for `sha256sum libc6_2.27-3ubuntu1.5_amd64.deb`?

krishnaadithya, 2022-03-28:
I have tried running docker_build_gpu.sh script several times it is the same error over and over again

What is the output for sha256sum libc6_2.27-3ubuntu1.5_amd64.deb?

cbc0dcc4803d2979d9b5e734ae2a4d45e20a1aaec8cfc895a2209285a9ff7573  libc6_2.27-3ubuntu1.5_amd64.deb

subhailams, 2022-03-28:
Any updates on this issue?? Facing same error while building GPU docker with Ubuntu 18

hanxue, 2022-03-28:
Hi @krishnaadithya ,

Per your original post

```
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/g/glibc/libc6_2.27-3ubuntu1.5_amd64.deb Hash Sum mismatch
Hashes of expected file:
- SHA512:84869a5cf0903b1045c7e7e58c6bfcf9800c40fdc8b651087fa448ea2af1c18158df29b92af6dd0113ea77e627a23d7a4da4bae683db6d33f8d9104f044f207f
- SHA256:c4af07dc8a2fdc9c4f25f103cd03bcca7231c19d9ac9171fa7eb5eecb7fc46d7
- SHA1:47e5405d0f9c7ff40daff7ec010ff816d901f35e [weak]
- MD5Sum:aef43cd3a8359820eb9a027c19a056c8 [weak]
- Filesize:2829792 [weak]
```

The SHA256 checksum for the deb package is `c4af07dc8a2fdc9c4f25f103cd03bcca7231c19d9ac9171fa7eb5eecb7fc46d7`

I tried it on an internal server and I got the correct result

```
$ wget http://archive.ubuntu.com/ubuntu/pool/main/g/glibc/libc6_2.27-3ubuntu1.5_amd64.deb
--2022-03-28 09:36:48--  http://archive.ubuntu.com/ubuntu/pool/main/g/glibc/libc6_2.27-3ubuntu1.5_amd64.deb
Resolving archive.ubuntu.com (archive.ubuntu.com)... 91.189.91.39, 91.189.91.38, 91.189.88.142, ...
Connecting to archive.ubuntu.com (archive.ubuntu.com)|91.189.91.39|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2829792 (2.7M) [application/x-debian-package]
Saving to: ‘libc6_2.27-3ubuntu1.5_amd64.deb’

100%[========================================>] 2,829,792   4.81MB/s   in 0.6s   

2022-03-28 09:36:49 (4.81 MB/s) - ‘libc6_2.27-3ubuntu1.5_amd64.deb’ saved [2829792/2829792]

$ sha256sum libc6_2.27-3ubuntu1.5_amd64.deb
c4af07dc8a2fdc9c4f25f103cd03bcca7231c19d9ac9171fa7eb5eecb7fc46d7  libc6_2.27-3ubuntu1.5_amd64.deb
```

The SHA256 checksum that you posted for the same file **does not** match. The value you posted is `cbc0dcc4803d2979d9b5e734ae2a4d45e20a1aaec8cfc895a2209285a9ff7573`

There is networking issue preventing your host from downloading the deb packages from archive.ubuntu.com correctly. 

Please can you try the `docker_build_gpu.sh` script from another machine?

If this problem keeps recurring, I will suggest raising a bug report with Ubuntu so that they are aware certain IP addresses are having problem with downloading.https://help.ubuntu.com/community/ReportingBugs

krishnaadithya, 2022-03-28:
@hanxue 

thank you for figuring out the root cause, ip was the issue, so I used a vpn to bypass it and it is downloading fine now.

@subhailams try using a vpn or your phone data.

@hanxue could you please take a look at this issue https://github.com/Xilinx/Vitis-AI/issues/732#issue-1183110995 ?

closed 2022-03-28 by krishnaadithya

krishnaadithya, 2022-03-31:
@hanxue 

Though the GPU docker worked, there is still an issue when I run the docker I get:
docker: Error response from daemon: could not select device driver "" with capabilities: [[gpu]]