Xilinx/Vitis-AI #735: Compiling models for DPUCVDX8H-2PE and 4PE

ksstms, 2022-03-29:
So, in #672 I learned that you can use the predefined [targets](https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/VART/target_factory/targets) to build models for the DPUCVDX8H. 

Up until now I didn't realize that they are only available for the [8PE](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Runtime/VART/target_factory/targets/DPUCVDX8H_ISA1_F2W2_8PE.prototxt) and [6PE](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Runtime/VART/target_factory/targets/DPUCVDX8H_ISA1_F2W4_6PE_aieDWC.prototxt) configurations. How can I build for the 4PE and 2PE variants?

SuZhan0322, 2022-09-01:
Hi @ksstms ,
Sorry for the late reply. The 4PE is released In VAI2.5, and you can compile the 4PE model directly.  
The 2PE is not released in VAI2.5, but you can refer to xo release and is compatible with 4PE's precompiled model.

SuZhan0322, 2022-09-29:
Hi @ksstms ,
Any update about this issue? Thanks!

ksstms, 2022-09-29:
Hi @SuZhan0322. Thank you, and I'm sorry for the late reply.
This part of our project was put on hold, as Vivado crashes while implementing the DPUCVDX8H on the VCK190 and I didn't get any help [on the forums](https://support.xilinx.com/s/question/0D52E000075ZUzMSAW/placement-abnormal-program-termination-11).
I'm glad that it's supported now. As far as I know VAI2.5 is for the 2022.1 version tools. Unfortunately we have an ES board, which is no longer supported in 2022.1. I guess we'll get back to this topic when our VCK190 production board finally arrives.

SuZhan0322, 2022-10-14:
No problem! @ksstms