Xilinx/Vitis-AI #737: Module without vitis_ai_library

China-CC-lon, 2022-03-30:
Regarding the resnet.py in the demo that failed to find the vitis_ai_library module, how can I find this module?
Code address: https://github.com/Xilinx/Vitis-AI/blob/v1.4.1/demo/Vitis-AI-Library/samples/graph_runner/resnet50_graph_runner_py/resnet50.py
or 
Regarding the python API described in the ug1414.pdf document, there is a program in creat_graph_runner as shown in the figure: 
[image]
The vitis_ai_library API cannot be found, what is the reason?
(The prerequisite for this is that the board has been installed, and it can quantify the module, compile the module, and get the xmodel I want)
Thank you in advance

qianglin-xlnx, 2022-03-30:
Hi @China-CC-lon 
Graph_runner API is based on Vitis-AI-Library. So you need to install it.  
For VAI2.0, it's libvitis_ai_library-2.0.0-r64.aarch64.rpm
You can find the library in: https://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc/VART/2021.2 
Note that it's based on 2021.2.

China-CC-lon, 2022-03-30:
> 嗨@China-CC-lon Graph_runner API 基于 Vitis-AI-Library。所以你需要安装它。 对于 VAI2.0，它是 libvitis_​​ai_library-2.0.0-r64.aarch64.rpm 您可以在以下位置找到该库：[https](https://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc/VART/2021.2) ://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc/VART/2021.2 注意它基于 2021.2。

Hi，qianglin-xlnx
OK, thanks for your help! I'll try it right away.

China-CC-lon, 2022-04-02:
Hi @qianglin-xlnx,  I am using VCK5000 in Ubuntu20.04 environment, according to the information you provided, it is an RPM file, do you have any deb file suitable for Ubuntu system?

qianglin-xlnx, 2022-04-02:
@wanghy-xlnx  Could you please help to provide the deb files for VCK5000?

@China-CC-lon You can also try to build the VART source files to generate the deb packages. 
Refer to: https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Runtime/VART/quick_start_for_edge.md