Xilinx/Vitis-AI #738: Can I check the weight of the model after compile?

SamsonChau, 2022-04-01:
Hi, I am working on a custom PyTorch model that already successful deployed on the zcu102 platform using vitis ai. However, the accuracy is way off before quantization and compile (Tested with same dataset). After reviewing the model before and after compile I notice that the difference may down to the weight of the model has change significantly after compile. 

Therefore, I wanna ask is there a way that I can review / check the weight of the xmodel after compile?  for now the only thing I can do is reviewing the compiled model using the xir tools or Netron to review the model.

wanghong4compiler, 2022-04-19:
Hi @SamsonChau ，
I'm sorry to say that it's difficult to check weights of the model after compilation. 
Considering to reduce the size of model, compiler removes all const data from the model, only keeps the ddr init segment.
In this segment, all const data in an int8-like format are concatenated together and may be shuffled for hardware convenient.

allen123wzh, 2022-05-04:
> Hi @SamsonChau ，
> I'm sorry to say that it's difficult to check weights of the model after compilation.
> Considering to reduce the size of model, compiler removes all const data from the model, only [...]

In this case is there any way to improve the model performance after compilation? In our scenario, the model accuracy after quantization is still good, but after passing the quantized model through the compiler, the performance degrades a lot like 10~30%. Vitis Ai 1.4 and PyTorch 1.7 were used, the board is ZCU102 and DPU is DPUCZDX8G, thanks.

wanghong4compiler, 2022-05-24:
Hi @allen123wzh ，
Would you mind sharing your model and dump the golden result, so that I can reproduce this issue?

qianglin-xlnx, 2022-06-23:
Hi @allen123wzh 
Has this issue been solved? If not, could you please share your model for our further analysis? Thank you.

allen123wzh, 2022-06-27:
> Hi @allen123wzh
> Has this issue been solved? If not, could you please share your model for our further analysis? Thank you.

Hi, sorry for the late reply. This issue has been fixed, it is our pre-processing problem instead of the compiler's fault. Thanks.

qianglin-xlnx, 2022-06-28:
@allen123wzh 
Glad to hear that you solved this issue.

closed 2022-06-28 by qianglin-xlnx