Xilinx/Vitis-AI #739: [VAIQ_ERROR]: /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/nn/_kernels.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZNK2at6Tensor8toStringEv

rohanpawar294, 2022-04-01:
I got this issue after building a new GPU docker image
[VAIQ_ERROR]: /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/nn/_kernels.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZNK2at6Tensor8toStringEv

shdalton, 2022-04-18:
Exactly same thing here... found any solution?

ydajwz123, 2022-05-09:
Exactly same thing here... found any solution?

niuxjxlnx, 2022-05-09:
Hi, all:

The error is because that vai_q_pytorch(pytorch_nndct) library is not compatible pytorch package in your docker. Once pytorch version is changed, vai_q_pytorch needs to be compiled again.

You can refer to the script /opt/vitis_ai/scripts/replace_pytorch.sh in vitis-ai 2.0 docker (Please search the script in user guide for some description).

ydajwz123, 2022-05-10:
> Hi, all:
>
> The error is because that vai_q_pytorch(pytorch_nndct) library is not compatible pytorch package in your docker. Once pytorch version is changed, vai_q_pytorch needs to be compiled again. [...]

THANKS for you reply. But I found that mamba install failed. What could be the problem.
```
Pinned packages:
  - python 3.6.*

Encountered problems while solving:
  - package torchaudio-0.9.0-py36 requires pytorch 1.9.0, but none of the providers can be installe
```

ydajwz123, 2022-05-10:
> @ydajwz123:
> 
> It seems that torchaudio you installed is not compatible with pytorch you installed. Please refer to this page to select the versions: https://pytorch.org/get-started/previous-versions/

I assume all the packages are equipped properly in the GPU-Docker. I just enter the docker environment and run the replace.sh. Should I reconfigure the environment before I run the replace.sh?

niuxjxlnx, 2022-05-12:
@ydajwz123 :

Which version of vitis-ai GPU docker are you used? Can you show the steps before your met the following problem?

[VAIQ_ERROR]: /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/nn/_kernels.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZNK2at6Tensor8toStringEv

DeepTrial, 2022-05-18:
I have met the same problem with error with vitis-ai 2.0 gpu docker
`[VAIQ_ERROR]: /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/nn/_kernels.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN6caffe28TypeMeta21_typeMetaDataInstanceIN3c107complexIfEEEEPKNS_6detail12TypeMetaDataEv`

janifer112x, 2022-09-05:
closed this issue , please refer tp issue#829 for the fix

closed 2022-09-05 by janifer112x