Xilinx/Vitis-AI #740: [UNILOG][FATAL][XCOM_ACGEN_ERROR][instuction generating fail, please contact us.]

mlopezpalma, 2022-04-03:
I am newer in vitis-ai ,  I am using  a ./docker_run.sh xilinx/vitis-ai-gpu:latest   in the tutorial example  08-tf2_flow

the  train  is working ok  ,  the  quantize  is working ok  also  

But in the complilation    source compile.sh zcu102

compile() {
      vai_c_tensorflow2 \
            --model           build/quant_model/q_model.h5 \
            --arch            $ARCH \
            --output_dir      build/compiled_$TARGET \
            --net_name        customcnn
}

Give the next error  

(vitis-ai-tensorflow2) Vitis-AI /workspace > source compile.sh zcu102
-----------------------------------------
COMPILING MODEL FOR ZCU102..
-----------------------------------------
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['build/quant_model/q_model.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/customcnn_org.xmodel', proto=None)
[INFO] tensorflow2 model: /workspace/build/quant_model/q_model.h5
[INFO] keras version: 2.6.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :100%|██████████| 71/71 [00:00<00:00, 21341.23it/s]               
[INFO] infer shape (NHWC)  :100%|██████████| 108/108 [00:00<00:00, 5136.93it/s]              
[INFO] perform level-0 opt :100%|██████████| 2/2 [00:00<00:00, 137.53it/s]                   
[INFO] perform level-1 opt :100%|██████████| 2/2 [00:00<00:00, 542.36it/s]                   
[INFO] generate xmodel     :100%|██████████| 108/108 [00:00<00:00, 4238.97it/s]              
[INFO] dump xmodel: /tmp/customcnn_org.xmodel
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Graph name: model, with op num: 232
[UNILOG][INFO] Begin to compile...
[UNILOG][FATAL][XCOM_ACGEN_ERROR][instuction generating fail, please contact us.] 
*** Check failure stack trace: ***
Aborted (core dumped)
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
-----------------------------------------
MODEL COMPILED
-----------------------------------------
(vitis-ai-tensorflow2) Vitis-AI /workspace > 

Similar problem happens  if  I use the   cpu  docker   instead of the   dpu  . I also test    with  ./docker_run.sh xilinx/vitis-ai:1.3.411  ( follow some  post  but  other errors  appears  )

I appreciate  any  help for this issue 

thanks

mlopezpalma, 2022-04-03:
Additional comment    if   the model is poorly trainned    with  2 epochs   accuracy   0.5   the  error  do not appear   ; but  if the model  is extensively  trained   with 250 epocs    accuracy  0.95   the error appear  .

wanghong4compiler, 2022-04-19:
Hi @mlopezpalma ，
would you mind upgrade the toolchain to VAI2.0? I test it on VAI2.0, found it compiled successfully.

wanghong4compiler, 2022-06-16:
Hi @AfifaIshtiaq ,
Is this one same with issue #854 ？

AfifaIshtiaq, 2022-06-16:
> Hi @AfifaIshtiaq ,
> Is this one same with issue #854 ？

Yes

qianglin-xlnx, 2022-06-23:
Same issue with #854

closed 2022-06-23 by qianglin-xlnx