Xilinx/Vitis-AI #743: Pre-built docker image xilinx/vitis-ai:2.0.0 issue

nhphuong91, 2022-04-05:
If anyone try to build docker base on the `xilinx/vitis-ai:2.0.0` docker image, you there will be any error at line `sudo apt-get update`. For example, build docker using docker file at `/external/onnxruntime/Dockerfile.vitis_ai_onnxrt`. The error is as below:
``` sh
W: GPG error: https://apt.kitware.com/ubuntu bionic InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 6AF7F09730B3F0A4
E: The repository 'https://apt.kitware.com/ubuntu bionic InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
A quick fix is to add `RUN apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 6AF7F09730B3F0A4` to docker file, before `sudo apt-get update` line

jzymessi, 2022-07-13:
@nhphuong91
    Hi , did you build onnxruntime docker success in Vitis-ai 2.5?

nhphuong91, 2022-07-14:
> @nhphuong91 Hi , did you build onnxruntime docker success in Vitis-ai 2.5?

@jzymessi I haven't tried it. But according to onnxe dockerfile in VAI 2.5, I believe this error would still exist unless Xilinx has updated the base image `xilinx/vitis-ai:2.0.0`

nhphuong91, 2022-07-20:
_**Update:**_ Now the key for nvidia repo expired also. You need to run these 2 commands before being able to use apt properly again:
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys A4B469963BF863CC
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 6AF7F09730B3F0A4