Xilinx/Vitis-AI #748: vaitrace error

ash8327, 2022-04-07:
Hi,

I am trying to use the vaitrace tool in AWS F1. I have set up the Vitis AI container correctly and I can run any model. But when I attempt to do:
 vaitrace --txt ./inception_example inception_v1_tf.xmodel ./image

Or similar on a model that works without vaitrace, I will get the error:

INFO:root:vaitrace time out, stopping process...
INFO:root:Generating ascii-table summary
ERROR:root:[]
ERROR:root:xmodel info raw info error
DPU Summary:
===================================================================================
DPU | Batch | SubGraph | Workload(GOP) | RunTime(ms) | Perf(GOP/s) | Mem(MB) | MB/s
----+-------+----------+---------------+-------------+-------------+---------+-----
===================================================================================

CPU Summary:
=============================================
Function | Device | Runs | AverageRunTime(ms)
---------+--------+------+-------------------
=============================================

I have tried changing the time in the command line with -t. I have also made sure I have root permission as instructed.

wanghy-xlnx, 2022-04-11:
Hi @ash8327 ,

Make sure you don't add sudo in front of the vaitrace command. That is, using the following command,
```
vaitrace --txt ./inception_example inception_v1_tf.xmodel ./image
```
instead of
```
sudo vaitrace --txt ./inception_example inception_v1_tf.xmodel ./image
```

If executed normally, the result should be similar to the following display.

[image]

If it doesn't work, please provide a screenshot of your completed log.

ash8327, 2022-05-02:
This is the error I am seeing with my model:

INFO:root:vaitrace timeout, waiting for Python thread terminated
INFO:root:Generating ascii-table summary
ERROR:root:[]
ERROR:root:xmodel info raw info error
DPU Summary:
===================================================================================
DPU | Batch | SubGraph | Workload(GOP) | RunTime(ms) | Perf(GOP/s) | Mem(MB) | MB/s
----+-------+----------+---------------+-------------+-------------+---------+-----
===================================================================================

CPU Summary:
=============================================
Function | Device | Runs | AverageRunTime(ms)
---------+--------+------+-------------------
=============================================
(vitis-ai-tensorflow2) Vitis-AI /workspace/starfish/src > vaitrace -p --txt app_mt.py

ash8327, 2022-05-02:
For the model you shared I get the same error:
`Image : ILSVRC2012_val_00000455.JPEG
top[0] prob = 0.991293  name = shoe shop, shoe-shop, shoe store
top[1] prob = 0.004051  name = cowboy boot
top[2] prob = 0.001689  name = clog, geta, patten, sabot
top[3] prob = 0.000427  name = Loafer
top[4] prob = 0.000427  name = butcher shop, meat market
INFO:root:vaitrace time out, stopping process...
INFO:root:Generating ascii-table summary
ERROR:root:[]
ERROR:root:xmodel info raw info error
DPU Summary:
===================================================================================
DPU | Batch | SubGraph | Workload(GOP) | RunTime(ms) | Perf(GOP/s) | Mem(MB) | MB/s
----+-------+----------+---------------+-------------+-------------+---------+-----
===================================================================================

CPU Summary:
=============================================
Function | Device | Runs | AverageRunTime(ms)
---------+--------+------+-------------------
=============================================`

jzymessi, 2022-08-09:
@wanghy-xlnx :
    I  get the same error when run vaitrace on vck5000 , does this error solved?
    Can i run it in Vitis-AI 2.0?

williamliao28, 2022-08-18:
I also get the same error when running `vaitrace` on AWS F1. Any update on this issue?

AlphaApodis, 2022-08-18:
The vaitrace tool is not available on the AWS F1 instance according to an AMD / Xilinx. I don't know if this is still true for Vitis-AI 2.5.