Xilinx/Vitis-AI #749: Request a u250 platform similar to the waa provided in issue #474

Temerson0, 2022-04-08:
Hello,
I have not been able to find any TRD for a U250 DPU based on DPUCADF8H. The u200 example for the waa resnet50 is a good baseline, but I need to deploy my app on the Azure cloud which uses a U250.

Can .xo file located at /Vitis-AI/dsa/WAA-TRD/accel/DPUV3INT8/dpdpuv3_wrapper.xo be used for a u250 waa build?

If not, can you provide one?

Thanks

sumitn-xilinx, 2022-04-08:
Hello. You can use the DPU TRD from here - https://github.com/Xilinx/Vitis-AI/tree/master/dsa/DPUCADF8H-XO
It has instructions for U250 as well.

Temerson0, 2022-04-08:
Thanks for the link. I forgot to specify that I am currently using Vitis AI 1.4.1. Do you know if the DPUCADF8H-XO can be used to build a custom application under Vitis AI 1.4.1?

Temerson0, 2022-04-11:
@sumitn-xilinx I was able to complete the build with some errors of the DPUCADF8H-XO for a U250 using platform xilinx_u250_gen3x16_xdma_3_1 and Vitis/Vivado 2021.2. It looks like two of the SLR builds (0 and 2) created errors, but the one for SLR 3 completed. I was then able to run the adas_detection demo with the xclbin created by the build. 

Here is the output of the errors from the U250 DPUCADF8H build:
`INFO: [v++ 60-1441] [20:41:53] Run run_link: Step xclbinutilinfo: Completed
Time (s): cpu = 00:00:00.99 ; elapsed = 00:00:01 . Memory (MB): peak = 2039.004 ; gain = 0.000 ; free physical = 104040 ; free virtual = 125309
INFO: [v++ 60-1443] [20:41:53] Run run_link: Step generate_sc_driver: Started
INFO: [v++ 60-1453] Command Line: 
INFO: [v++ 60-1454] Run Directory: /data1/Software/Xilinx/Vitis-AI/dsa/DPUCADF8H-XO/projects/ml_shell/build_dir.hw.xilinx_u250_gen3x16_xdma_3_1_202020_1/link/run_link
INFO: [v++ 60-1441] [20:41:53] Run run_link: Step generate_sc_driver: Completed
Time (s): cpu = 00:00:00 ; elapsed = 00:00:00.01 . Memory (MB): peak = 2039.004 ; gain = 0.000 ; free physical = 104040 ; free virtual = 125309
ERROR: [v++ 60-2334] Compute unit DPUCADF8H_1 appears to span more than one SLR. SLR0 was specified.
ERROR: [v++ 60-2334] Compute unit DPUCADF8H_2 appears to span more than one SLR. SLR2 was specified.
INFO: [v++ 60-2331] SLR3 was specfied for compute unit DPUCADF8H_3, and verified as such in implementation.
INFO: [v++ 60-244] Generating system estimate report...`

Do you have any suggestions to keep DPU_1 and DPU_2 confined to a single SLR? I was thinking of using --optimize 3, but would like to get your input.

vgudur-amd, 2022-07-27:
Hi @Temerson0,
Are you still facing this issue? If not we can close the issue.

Temerson0, 2022-07-29:
@vgudur-amd I never received a response from my last question. Have you resolved the build issue with the U250 for the issue request?
https://github.com/Xilinx/Vitis-AI/tree/master/dsa/DPUCADF8H-XO
The above link completed building with errors on 1.4.1 as described above. 
What versions of Vitis-AI and Vitis has this DPU TRD been tested on?

vgudur-amd, 2022-08-01:
Hi @Temerson0 
Any specific reason to stick with 1.4.1? The link you shared above for dsa area doesn't work for VAI2.5. I don't see U250 related setup in 1.4.1 branch as well. However, it is there in 2.0. Since, it is released in VAI2.0, and if there is no restriction on VAI version, may I suggest you to use VAI2.0 DPU files?

Temerson0, 2022-08-03:
@vgudur-amd I have tried to use VAI2.0, but have run into many problems running the examples for the U200 and U250, that is why I have kept using VAI1.4.1. At this time, the instability and my lack of success with using Vitis AI for my specific object detection project is pushing me toward using a GPU.

vgudur-amd, 2022-08-10:
Hi @Temerson0,
If your system doesn't require more than one DPU, you may change the setup and constraints files to remove unwanted DPU/s. In VAI2.0, hope you have followed all the setup related instructions for the Alveo cards. In WAA, u200 example was working fine in VAI2.0 release.