Xilinx/Vitis-AI #752: Kernel Panic when porting WAA ADAS example on ZCU104

haipnh, 2022-04-08:
Hi, 

I am trying to make dsa/WAA-TRD/proj/build/detection-pre_DPUv2 running on my ZCU104.

My environment:

- Host: Ubuntu 18.04.4 / Vitis 2020.2 / Vitis-AI 1.4.1
- Target: ZCU104 / xilinx_zcu104_base_202020_1.xpfm / xilinx-zynqmp-common-v2020.2

I followed the instruction and make the changes as following:

``` shell
diff --git a/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/build_detection_pre.sh b/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/build_detection_pre.sh
index e3652b946..6eb9ea60f 100755
--- a/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/build_detection_pre.sh
+++ b/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/build_detection_pre.sh
@@ -1,9 +1,12 @@
+export EDGE_COMMON_SW=/opt/xilinx/images/xilinx-zynqmp-common-v2020.2
+export SDX_PLATFORM=/opt/xilinx/platforms/xilinx_zcu104_base_202020_1/xilinx_zcu104_base_202020_1.xpfm
+
 #Generate Pre-processor pipeline xo and so file
 export CUR_DIR=$PWD
 cd $CUR_DIR/../../../accel/detection-pre
-make cleanall
+#make cleanall
 make xo TARGET=hw BOARD=Zynq ARCH=aarch64
 
 #Generate SD card files
 cd $CUR_DIR
-make KERNEL=DPU_PP PP_ACCEL=../../../accel/detection-pre DPU_IP=../../../../DPU-TRD/dpu_ip/ DEVICE=zcu102
+make KERNEL=DPU_PP PP_ACCEL=../../../accel/detection-pre DPU_IP=../../../../DPU-TRD/dpu_ip/ DEVICE=zcu104
diff --git a/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/config_file/prj_config b/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/config_file/prj_config
index 9083fd5da..29700e452 100755
--- a/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/config_file/prj_config
+++ b/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/config_file/prj_config
@@ -19,20 +19,20 @@
 
 freqHz=300000000:DPUCZDX8G_1.aclk
 freqHz=600000000:DPUCZDX8G_1.ap_clk_2
-freqHz=300000000:DPUCZDX8G_2.aclk
-freqHz=600000000:DPUCZDX8G_2.ap_clk_2
+#freqHz=300000000:DPUCZDX8G_2.aclk
+#freqHz=600000000:DPUCZDX8G_2.ap_clk_2
 
 [connectivity]
 
 sp=DPUCZDX8G_1.M_AXI_GP0:HPC0
 sp=DPUCZDX8G_1.M_AXI_HP0:HP0
 sp=DPUCZDX8G_1.M_AXI_HP2:HP1
-sp=DPUCZDX8G_2.M_AXI_GP0:HPC0
-sp=DPUCZDX8G_2.M_AXI_HP0:HP2
-sp=DPUCZDX8G_2.M_AXI_HP2:HP3
+#sp=DPUCZDX8G_2.M_AXI_GP0:HPC0
+#sp=DPUCZDX8G_2.M_AXI_HP0:HP2
+#sp=DPUCZDX8G_2.M_AXI_HP2:HP3
 
 
-nk=DPUCZDX8G:2
+nk=DPUCZDX8G:1
 
 [advanced]
 misc=:solution_name=link
@@ -41,6 +41,6 @@ misc=:solution_name=link
 #param=compiler.skipTimingCheckAndFrequencyScaling=1
 
 [vivado]
-prop=run.impl_1.strategy=Performance_Explore
+prop=run.impl_1.strategy=Performance_ExploreWithRemap
 #param=place.runPartPlacer=0
```

After flashing the image onto a SD card and installing required packages, I ran the example by running commands:
```
root@zynqmp-common-2020_2:/media/sd-mmcblk0p1/adas_detection# ./adas_detection model_zcu104/yolov3_adas_pruned_0_9/yolov3_adas_pruned_0_9.xmodel 1
```

The Serial terminal dumped these information and system hanged:
```
root@zynqmp-common-2020_2:~# [ 4995.854297] Internal error: Oops: 96000144 [#1] SMP
[ 4995.859166] Modules linked in: zocl(O) dmaproxy(O) mali(O) [last unloaded: zocl]
[ 4995.866556] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           O      5.4.0-xilinx-v2020.2 #1
[ 4995.875067] Hardware name: ZynqMP ZCU104 RevC (DT)
[ 4995.879843] pstate: 40000005 (nZcv daif -PAN -UAO)
[ 4995.884622] pc : __dma_inv_area+0x40/0x58
[ 4995.888618] lr : arch_sync_dma_for_cpu+0x24/0x30
[ 4995.893216] sp : ffff800010003d10
[ 4995.896515] x29: ffff800010003d10 x28: ffff000878f34c00 
[ 4995.901810] x27: 0000000080c0c03c x26: ffff800011469508 
[ 4995.907104] x25: ffff000878ca0800 x24: ffff000878ca07c0 
[ 4995.912399] x23: 0000000000000002 x22: 0000000000000002 
[ 4995.917694] x21: 0000000000000600 x20: 5f74666965dabcc0 
[ 4995.922989] x19: ffff00087ab2f410 x18: 000000000000000e 
[ 4995.928283] x17: 0000000000000001 x16: 0000000000000019 
[ 4995.933578] x15: 0000000000000033 x14: 000000000000004c 
[ 4995.938873] x13: 0000437fd3591cd0 x12: 0000000000000000 
[ 4995.944168] x11: 0000000000000003 x10: 0000000000000000 
[ 4995.949463] x9 : 0000000000000000 x8 : ffff00087f781200 
[ 4995.954757] x7 : 0000000000000000 x6 : ffff000865dabcc0 
[ 4995.960052] x5 : 000000000000003c x4 : 5f74666965dabcc0 
[ 4995.965347] x3 : 000000000000003f x2 : 0000000000000040 
[ 4995.970642] x1 : 5f73666965dac2c0 x0 : 5f73666965dabcc0 
[ 4995.975937] Call trace:
[ 4995.978370]  __dma_inv_area+0x40/0x58
[ 4995.982016]  dma_direct_sync_single_for_cpu+0x7c/0x80
[ 4995.987057]  dma_direct_unmap_page+0x84/0x88
[ 4995.991313]  gem_rx+0xf8/0x2b8
[ 4995.994357]  macb_poll+0x58/0x100
[ 4995.997658]  net_rx_action+0x114/0x340
[ 4996.001397]  __do_softirq+0x118/0x22c
[ 4996.005043]  irq_exit+0x98/0xc0
[ 4996.008169]  __handle_domain_irq+0x64/0xb8
[ 4996.012246]  gic_handle_irq+0x5c/0xb8
[ 4996.015893]  el1_irq+0xb8/0x140
[ 4996.019018]  arch_cpu_idle+0x10/0x18
[ 4996.022577]  do_idle+0x200/0x280
[ 4996.025788]  cpu_startup_entry+0x20/0x28
[ 4996.029695]  rest_init+0xb0/0xbc
[ 4996.032906]  arch_call_rest_init+0xc/0x14
[ 4996.036898]  start_kernel+0x400/0x42c
[ 4996.040547] Code: 8a230000 54000060 d50b7e20 14000002 (d5087620) 
[ 4996.046634] ---[ end trace b57b578f9e38743e ]---
[ 4996.051237] Kernel panic - not syncing: Fatal exception in interrupt
[ 4996.057574] SMP: stopping secondary CPUs
[ 4996.061480] Kernel Offset: disabled
[ 4996.064951] CPU features: 0x0002,20002004
[ 4996.068943] Memory Limit: none
[ 4996.071984] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```
I have no idea with this situation and appreciate any useful information to overcome this issue.

sumitn-xilinx, 2022-04-08:
can you please follow the steps from the latest Vitis AI release instead of 1.4.1?
https://github.com/Xilinx/Vitis-AI/tree/master/Whole-App-Acceleration/apps/resnet50

mahadurkarm, 2022-04-08:
Kernel panic issue can be cause by incorrect kernel interface. I think your xo file is not properly generated. So please don't remove "make cleanall" command from build_detection_pre.sh. Also prj_config file is updated to use 1 instance of DPU kernel instead of 2 instead. It is recommended to use 2 inst of DPU kernel.

haipnh, 2022-04-08:
Thanks @mahadurkarm , I uncommented "make cleanall" and rebuilt it, still not working.
However, is there any proof for changing the number of DPU cores will cause problem? 
Logically I see it only affects the performance.
I tested a single core DPU-TRD on both ZCU104 and PYNQ-ZU using PYNQ-v2.7. It can run all the examples. FYI.

haipnh, 2022-04-09:
Thanks @sumitn-xilinx ,
I modified the adas_detection example on tag v2.0, it can run on ZCU104. I will use this to study the application architecture.

```
./app_test.sh --xmodel_file model_zcu104/yolov3_adas_pruned_0_9/yolov3_adas_pruned_0_9.xmodel --image_dir ./img/ --performance_diff

 Running Performance Diff: 

   Running Application with Software Preprocessing 

   E2E Performance: 14.19 fps
   Pre-process Latency: 21.61 ms
   Execution Latency: 8.41 ms
   Post-process Latency: 40.42 ms

   Running Application with Hardware Preprocessing 

   E2E Performance: 27.30 fps
   Pre-process Latency: 0.77 ms
   Execution Latency: 8.26 ms
   Post-process Latency: 27.58 ms

   The percentage improvement in throughput is 92.39 %
```

closed 2022-04-09 by haipnh