Xilinx/Vitis-AI #755: Execution of build for Xilinx WAA-TRD classification-pre_DPUv3int8 fails

Temerson0, 2022-04-11:
Hello,

I am encountering a problem when building a DPUCADF8H for a U200 with a classification pre-processor. 
The following instructions are being followed:
https://github.com/Xilinx/Vitis-AI/tree/1.4.1/dsa/WAA-TRD/proj/build/classification-pre_DPUv3int8

**Successful:**
The application runs correctly when I use the Xilinx provided DPU with HW preprocessor xclbin at:
 `/opt/xilinx/overlaybins/DPUCADF8H/waa/classification/dpuv3int8_pre.hw.xilinx_u200_gen3x16_xdma_1_202110_1.xclbin`

Firmware, app, model:
```
XLNX_VART_FIRMWARE=/opt/xilinx/overlaybins/DPUCADF8H/waa/classification/dpuv3int8_pre.hw.xilinx_u200_gen3x16_xdma_1_202110_1.xclbin
App: dsa/WAA-TRD/app/resnet50_int8
Model: resnet_v1_50_tf
```

**Fails:**
The following custom build of two DPUS with a HW preprocessor completes successfully but fails when running the same app as described above:

Custom build location:
[`dsa/WAA-TRD/proj/build/classification-pre_DPUv3int8`

Firmware, app, model:
```
XLNX_VART_FIRMWARE=/workspace/dsa/WAA-TRD/proj/build/classification-pre_DPUv3int8/outputs/xclbin/dpdpuv3_wrapper.hw.xilinx_u200_gen3x16_xdma_1_202110_1.xclbin
App: dsa/WAA-TRD/app/resnet50_int8
Model: resnet_v1_50_tf
```

Kernels
SLR0: DPUCADF8H_1
SLR1: pp_pipeline_accel_1
SLR2: DPUCADF8H_2

Follows is the code that causes the segmentation fault and a dump of the gdb session. It seg faults at xir::XrtBinStream::init_cu_names(). The names from the build appear to be correct. See the IP_LAYOUT.json file that is displayed below. It looks good, so I have no idea why the following code is failing with the segmentation fault. 
**Please provide guidance and how to correct this problem.** 
Note: that I have not had success running any of the demos on a U200 using Vitis 2.0, so that is why I am using Vitis 1.4.1.

**File:** `tools/Vitis-AI-Runtime/VART/vart > vi xrt-device-handle/src/xrt_bin_stream.cpp`

```
void XrtBinStream::init_cu_names() {
  for (auto i = 0; i < ip_layout_->m_count; ++i) {
    if (ip_layout_->m_ip_data[i].m_type != IP_KERNEL) continue;
    cu_names_.push_back(std::string((char*)ip_layout_->m_ip_data[i].m_name));
  }
}
```
GDB session:

```
gdb --args ./resnet50_int8 resnet_v1_50_tf/resnet_v1_50_tf.xmodel /workspace/CK-TOOLS/dataset-imagenet-ilsvrc2012-val-min 0 0
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law. Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./resnet50_int8...done.

(gdb) run
Starting program: /workspace/dsa/WAA-TRD/app/resnet50_int8/resnet50_int8 resnet_v1_50_tf/resnet_v1_50_tf.xmodel /workspace/CK-TOOLS/dataset-imagenet-ilsvrc2012-val-min 0 0
warning: Error disabling address space randomization: Operation not permitted
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0408 15:47:18.918046 6474 [main.cc](https://main.cc/):362] create running for subgraph: subgraph_resnet_v1_50/block1/unit_1/bottleneck_v1/add
Program received signal SIGSEGV, Segmentation fault.
0x00007f54d41c8f97 in xir::XrtBinStream::init_cu_names() ()
  from /usr/lib/[librt-engine.so](https://librt-engine.so/)

(gdb) where
#0 0x00007f54d41c8f97 in xir::XrtBinStream::init_cu_names() ()
  from /usr/lib/[librt-engine.so](https://librt-engine.so/)
#1 0x00007f54d41c91dd in xir::XrtBinStream::XrtBinStream(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
  from /usr/lib/[librt-engine.so](https://librt-engine.so/)
#2 0x00007f54d414ee4c in XrmResource::alloc_from_attrs(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, char*, xir::Attrs*) () from /usr/lib/[librt-engine.so](https://librt-engine.so/)
#3 0x00007f54d41500f5 in XrmResource::XrmResource(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xir::Attrs*) ()
  from /usr/lib/[librt-engine.so](https://librt-engine.so/)
#4 0x00007f54d415169b in DeviceHandle::DeviceHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xir::Attrs*) ()
  from /usr/lib/[librt-engine.so](https://librt-engine.so/)
#5 0x00007f54d415241b in XrtDeviceHandle::XrtDeviceHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xir::Attrs*) () from /usr/lib/[librt-engine.so](https://librt-engine.so/)
#6 0x00007f54d4196a6f in XclDpuController<XrtDeviceHandle, XrtDeviceBuffer, XrtDeviceBuffer>::XclDpuController(xir::Subgraph const*, xir::Attrs*) ()
  from /usr/lib/[librt-engine.so](https://librt-engine.so/)
#7 0x00007f54d419f65b in Dpuv3Int8Controller::Dpuv3Int8Controller(xir::Subgraph const*) () from /usr/lib/[librt-engine.so](https://librt-engine.so/)
#8 0x00007f54d4149207 in std::shared_ptr<DpuController> DpuControllerFactory::get<xir::Subgraph const*>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, xir::Subgraph const*, xir::Attrs*) ()
  from /usr/lib/[librt-engine.so](https://librt-engine.so/)
#9 0x00007f54d414b7ee in vart::DpuRunner::DpuRunner(xir::Subgraph const*)
  () from /usr/lib/[librt-engine.so](https://librt-engine.so/)
#10 0x00007f54d414bd92 in create_runner () from /usr/lib/[librt-engine.so](https://librt-engine.so/)

#11 0x00007f54ecaeb55d in vart::Runner::create_runner(xir::Subgraph const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /usr/lib/libvart-runner.so.1
#12 0x0000563efa16ed7f in main () at src/[main.cc](https://main.cc/):364
#13 0x00007f54ea7f6bf7 in __libc_start_main ()
---Type <return> to continue, or q <return> to quit---:q
  from /lib/x86_64-linux-gnu/libc.so.6
#14 0x0000563efa16f10a in _start () at src/[main.cc](https://main.cc/):369
```
**IP_LAYOUT.json**: (Head of file)
```
{
    "ip_layout": {
        "m_count": "10",
        "m_ip_data": [
            {
                "m_type": "IP_KERNEL",
                "m_int_enable": "1",
                "m_interrupt_id": "1",
                "m_ip_control": "AP_CTRL_CHAIN",
                "m_base_address": "0xc10000",
                "m_name": "DPUCADF8H:DPUCADF8H_1"
            },
            {
                "m_type": "IP_KERNEL",
                "m_int_enable": "1",
                "m_interrupt_id": "0",
                "m_ip_control": "AP_CTRL_CHAIN",
                "m_base_address": "0x1410000",
                "m_name": "DPUCADF8H:DPUCADF8H_2"
            },
            {
                "m_type": "IP_KERNEL",
                "m_int_enable": "1",
                "m_interrupt_id": "2",
                "m_ip_control": "AP_CTRL_CHAIN",
                "m_base_address": "0x1010000",
                "m_name": "pp_pipeline_accel:pp_pipeline_accel_1"
            },
...
```

vgudur-amd, 2022-07-27:
Hi @Temerson0,
Are you still facing this issue? If not we can close the issue.

Temerson0, 2022-07-28:
@vgudur-amd As far as I know it still fails under 1.4. Have you tested it on your end and on which versions of Vitis AI?

vgudur-amd, 2022-08-01:
Hi @Temerson0,
I hope you are using correct VART libraires for the release version you are using. Also, could you please check the app with placing your custom xclbin in the same overlaybins path and setting the XLNX_VART_FIRMWARE appropriately? 
FYI, VAI2.0 has build flow including pre-process accelerator. The app is also unified for all the platforms in the VAI2.0.

Temerson0, 2022-08-03:
The libraries and paths are set correctly, see below. I have had problems running the examples on a U200 with VAI2.0, that is why I stayed with VAI1.4.1. Is there a later version that has been tested with either a U200 or U250? 

ls /usr/lib/*vart-runner*
/usr/lib/libvart-runner-assistant.so        /usr/lib/libvart-runner.so
/usr/lib/libvart-runner-assistant.so.1      /usr/lib/libvart-runner.so.1
/usr/lib/libvart-runner-assistant.so.1.4.1  /usr/lib/libvart-runner.so.1.4.1

ls /usr/lib/*vart-util*
/usr/lib/libvart-util.so    /usr/lib/libvart-util.so.1.4.1
/usr/lib/libvart-util.so.1

 echo $XLNX_VART_FIRMWARE
/workspace/dsa/WAA-TRD/proj/build/classification-pre_DPUv3int8/outputs/xclbin/dpdpuv3_wrapper.hw.xilinx_u200_gen3x16_xdma_1_202110_1.xclbin

echo $XCLBIN_PATH
/workspace/dsa/WAA-TRD/proj/build/classification-pre_DPUv3int8/outputs/xclbin/

vgudur-amd, 2022-08-10:
Please try placing the xclbin in the overlaybins path as given in the example. Also, in the custom build, I hope you are following the cu names correctly. You may need to modify the app code depending on the cu names.

qianglin-xlnx, 2023-04-22:
Closing since no activity for more than half year, please open a new issue if you still have any questions. Thanks.

closed 2023-04-22 by qianglin-xlnx