Xilinx/Vitis-AI #760: Cannot find runner attr, this subgraph may not be compiled!

TrajM, 2022-04-15:
Hello,

I have a model that I compiled and deployed without issues when using the VART python API. I am switching to the C++ API and I keep running into an error.

My model has 3 subgraphs. With the python API, I use this method to assign my DPU runners:
```   
   for sub in subgraphs:
        name=sub.get_name()
        if name.find("I1") != -1: 
            runner1=vart.Runner.create_runner(sub, "run")
        elif name.find("I2") != -1:
            runners2=vart.Runner.create_runner(sub, "run")
        elif name.find("I3") != -1:
            runner3=vart.Runner.create_runner(sub, "run")
 ```
and a call my runners in the order I scheduled without problems.

I wrote my C++ code this way:
```  
    auto root = graph->get_root_subgraph();
    xir::Subgraph* subgraph1 = nullptr;
    xir::Subgraph* subgraph2 = nullptr;
    xir::Subgraph* subgraph3 = nullptr;
    for (auto c : root->children_topological_sort()) {
      if ((c->get_name().find("I1")!= string::npos) && subgraph1 == nullptr) {
        subgraph1 = c;
      }
      if ((c->get_name().find("I2")!= string::npos) && subgraph2 == nullptr) {
        subgraph2 = c;
      }
     if ((c->get_name().find("I3")!= string::npos) && subgraph3 == nullptr) {
        subgraph3 = c;
      }
      auto runner1 = vart::Runner::create_runner(subgraph1, "run");
      auto runner2 = vart::Runner::create_runner(subgraph2, "run");
      auto runner3 = vart::Runner::create_runner(subgraph3, "run");    
```  
  After compiling and running the code, I get the following message:
  `F0415 09:36:34.877319  1970 dpu_runner.cpp:146] [UNILOG][FATAL][VART_RUNNER_CONSTRUCTION_FAIL][Cannot create runner] Cannot find runner attr, this subgraph may not be compiled! subgraph name: subgraph_I1`
  
  Since the xmodel file works already with python, I don't understand why it says that it is not compiled when called with C++.
  (I am using Vitis AI 1.3.1)

wilderfield, 2022-04-15:
Just a wild guess…

I notice when you iterate across subgraphs in C++ you are implicitly making a copy with

auto c

c only exists in the context of your for loop.

I think if you switch to

auto &c

Your code will work.

TrajM, 2022-04-15:
Thanks for the hint!
I copied the code from one of the sample codes of VART in the repo and my C++ is a bit rusty so I am a bit lost with all the pointers.

I will try your advice and see if it is solved.
Thanks

TrajM, 2022-04-18:
Unfortunately, this approach did not solve the issue.
I am following the code in 
[resnet50_zero_copy/resnet50_b.cpp](https://github.com/Xilinx/Vitis-AI/blob/1.3.1/tools/Vitis-AI-Runtime/VART/vart/dpu-runner/samples/resnet50_zero_copy/resnet50_b.cpp)

 I am following the approach in lines 100-113
```
  auto xmodel_file = std::string(argv[1]);
  const auto image_file_name = std::string(argv[2]);
  {
    auto graph = xir::Graph::deserialize(xmodel_file);
    auto root = graph->get_root_subgraph();
    xir::Subgraph* subgraph = nullptr;
    for (auto c : root->children_topological_sort()) {
      if (c->get_attr<std::string>("device") == "DPU" && subgraph == nullptr) {
        subgraph = c;
      }
    }
    auto attrs = xir::Attrs::create();
    std::unique_ptr<vart::RunnerExt> runner =
        vart::RunnerExt::create_runner(subgraph, attrs.get());
```

Is there a complete documentation of the XIR graphs?
I only found
https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/VART/xir

and

https://docs.xilinx.com/r/1.3-English/ug1414-vitis-ai/XIR

but they are too general, I need to understand more the output types of the subgraph functions.

Thanks

TrajM, 2022-04-18:
I figured out the issue, the code is fine but the graph itself has multiple subgraphs including the same name "I1", "I2" or "I3".
For example, "subgraph_I1" here, which I assume is just the input layer, is not compiled for DPU, I should look for "subgraph_quant_I1_conv1" and assign it to the DPU. Same goes for the rest of subgraphs.

[image]

I will close this issue, but a better XIR documentation would be very helpful in the future.

closed 2022-04-18 by TrajM