Xilinx/Vitis-AI #761: PyTorch Quantization - [VAIQ_WARN]: Node ouptut tensor is not quantized

Orchidaceae, 2022-04-17:
I'm trying to quantize and export a simple 2 layer MLP, code seen below:

```
class Feedforward(torch.nn.Module):
    def __init__(self, input_size, hidden_size):
        super(Feedforward, self).__init__()
        self.input_size = input_size
        self.hidden_size = hidden_size
        self.fc1 = torch.nn.Linear(self.input_size, self.hidden_size)
        self.relu = torch.nn.ReLU()
        self.fc2 = torch.nn.Linear(self.hidden_size, 1)
        self.sigmoid = torch.nn.Sigmoid()

    def forward(self, x):
        hidden = self.fc1(x)
        relu = self.relu(hidden)
        output = self.fc2(relu)
        output = self.sigmoid(output)
        return output
```

## I use the following code to quantize and export my model:

```
model = load_model()
input = torch.randn([2,2,2])

model.eval()

quantizer = torch_quantizer('calib', model, (input))
quant_model = quantizer.quant_model

quantizer.export_quant_config()

input = torch.randn([1,2,2])
quantizer = torch_quantizer('test', model, (input))

quantizer.export_xmodel()
```

## I get the following warning when I run  ```export_quant_config()```:

```[VAIQ_WARN]: Node ouptut tensor is not quantized: Feedforward::input_0 type: input```
and the same warning for all layers and parameters in my model.

### When I debugg the function I can gather the following quantization info about my model:
```
> (Pdb) TORCHQuantizer._QuantInfo 
> {'param': {'Feedforward::fc1.weight': [8, None], 'Feedforward::fc1.bias': [8, None], 'Feedforward::fc2.weight': [8, None], 'Feedforward::fc2.bias': [8, None]}, 'output': {'Feedforward::input_0': [8, None], 'Feedforward::Feedforward/ReLU[relu]/input': [8, None], 'Feedforward::Feedforward/Linear[fc2]/51': [8, None]}, 'input': {}}
```
I' m guessing that the *bnfp* value should not be None but I have no Idea how to properly quantize my model. 

**Any suggestions?**

DouDinai, 2022-04-22:
I have the same question
 [VAIQ_WARN]: Node ouptut tensor is not quantized: YoloBody::input_0 type: input
[VAIQ_WARN]: Node ouptut tensor is not quantized: YoloBody::YoloBody/Focus[backbone]/CSPDarknet[backbone]/Focus[stem]/input.1 type: concat

[VAIQ_WARN]: Node ouptut tensor is not quantized: YoloBody::YoloBody/Focus[backbone]/CSPDarknet[backbone]/Focus[stem]/BaseConv[conv]/Conv2d[conv]/input.2 type: conv2d

[VAIQ_WARN]: Node ouptut tensor is not quantized: YoloBody::YoloBody/Focus[backbone]/CSPDarknet[backbone]/Focus[stem]/BaseConv[conv]/SiLU[act]/input.3 type: elemwise_mul
……
And the final tip is 
[VAIQ_WARN]: Quantization is not performed completely, check if model inference function is called!!!

zl200881, 2022-04-25:
Model forward loop for evaluation is needed before export quantization config.
Please refer to the function "evaluate" in the demo resnet18_quant.py.

zl200881, 2022-04-25:
Model forward loop for evaluation is needed before exporting quantization config.
Please refer to the function "evaluate" in the example resnet18_quant.py.

closed 2022-05-18 by niuxjxlnx