Xilinx/Vitis-AI #762: Does vck5000 support running models with multiple batch sizes?

jia0511, 2022-04-19:
Hello,
From page 64 of your official document UG1414, the model can only support batchsize=1 when quantized to xmodel by pytorch, which means that it can only run in dpu with a single batchsize. We observed that a single batchsize cannot reach the peak computing power, If I want to improve the utilization efficiency of dpu, what changes do I need to make? Do you plan to support multiple batch sizes in the future?

wanghy-xlnx, 2022-04-26:
Hi @jia0511 ,
The model compiled by the pytorch model can support batch_size>1. I don't see the description on page 64, can you confirm which chapter it is in?

jzymessi, 2022-04-27:
> 
[https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Running-Quantization-and-Getting-the-Result](url)
[image]

jia0511, 2022-05-02:
Dear all,
I modified the configuration parameters of music quantization and found that the pytorch tool can quantify batchsizes greater than 1, so I closed this issue.

jia0511, 2022-05-06:
Dear  wanghy-xlnx,

When the batch size of the model after quantization is 4, when I use vitis-ai pytorch vai_c_xir to compile the xmodel model, the following error is displayed. From the error message, is it currently not supported to compile models with batch size greater than 1 into the DPU?

> vai_c_xir -x reid_resnet50/Baseline_int.xmodel -a /workspace/platform_arch/arch_vck5000.json  -o output

[error.log](https://github.com/Xilinx/Vitis-AI/files/8638097/error.log)

> Hi @jia0511 , The model compiled by the pytorch model can support batch_size>1. I don't see the description on page 64, can you confirm which chapter it is in?

wanghy-xlnx, 2022-05-08:
Hi @jia0511 ,

We confirmed that:

For all DPU IP(except DPUCADF8H), vai_c_xir prefers the batch to be 1, and runtime will handle the multi-batch configuration according the IP's real batch number. If batchsize>1 when quantize model, although no error, but many OPs may be assigned to the CPU for execution.

In addition, I see from the log that the DPU you are using is DPUCZDX8G_ISA0_B4096_MAX. Please check that you are using the correct DPU and support multiple batches. Or use precompiled xclbin for testing according [README](https://github.com/Xilinx/Vitis-AI/tree/master/setup/vck5000).

jia0511, 2022-05-16:
Hi, @wanghy-xlnx 

Sorry, I may have sent the wrong log, this is the correct log, which means that the vck5000 hardware does not support it?

[vck5000_6pe_error.log](https://github.com/Xilinx/Vitis-AI/files/8698865/vck5000_6pe_error.l
[vck5000_8pe_error.log](https://github.com/Xilinx/Vitis-AI/files/8698866/vck5000_8pe_error.log)
og)

wanghy-xlnx, 2022-05-17:
Hi @jia0511 ,

Yes, this DPU is correct, but you still need to use batchsize=1 when compiling xmodel.

When actually running, runtime will handle the batchsize>1 configuration according to the DPU's real batch size.

jia0511, 2022-05-17:
@wanghy-xlnx 

Thank you for your answer, it is now clear that there is another question: is there an upper limit on the batch size at runtime, and whether the batch size is too large will cause the out of memory problem？

wanghy-xlnx, 2022-05-17:
Hi @jia0511 ,

It depends on hardware resources and model size. For 8PE, means the bachsize=8, and for 6PE, means the bachsize=6.