Xilinx/Vitis-AI #764: TF2 Quantizer bug in vitis_8_bit_quantize_transforms.py

aimilefth, 2022-04-20:
I came across this bug while using Vitis AI 2.0 to quantize a model.
<pre>Traceback (most recent call last):
  File &quot;quantize.py&quot;, line 136, in &lt;module&gt;
    main()
  File &quot;quantize.py&quot;, line 132, in main
    quant_model(args.float_model, args.quant_model, args.batchsize, args.tfrec_dir, args.evaluate)
  File &quot;quantize.py&quot;, line 73, in quant_model
    quantized_model = quantizer.quantize_model(calib_dataset=quant_dataset)
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py&quot;, line 578, in quantize_model
    calib_steps)
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py&quot;, line 271, in _calibrate_without_loss
    mode=&apos;QCB&apos;)
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py&quot;, line 745, in create_quantize_model
    quantize_strategy, mode)
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py&quot;, line 802, in quantize_apply
    quantize_strategy.get_quantize_registry(), mode)
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py&quot;, line 204, in apply
    layer_metadata).recursive_transform()
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py&quot;, line 738, in recursive_transform
    self.layer_metadata).transform()
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py&quot;, line 674, in transform
    copy.deepcopy(match_layer_node))
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_quantize_transforms.py&quot;, line 188, in replacement
    logger.info(info_msg.format(layer_node[&apos;name&apos;], activation, ug_link))
KeyError: &apos;name&apos;</pre>

So i opened the file 
/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/**vitis_8bit_quantize_transforms.py**
**and changed the line 188 from**

logger.info(info_msg.format(**layer_node['name']**, activation, ug_link))
to
logger.info(info_msg.format(**layer_node['config']['name'],** activation, ug_link))

and it works

aimilefth, 2022-04-20:
Moving on to the Compile, i encountered a similar error
<pre>[INFO] Tensorflow Keras model type: sequential
[INFO] parse raw model     :  0%|          | 0/18 [00:00&lt;?, ?it/s]              [INFO] parse raw model     : 17%|█▋        | 3/18 [00:00&lt;00:00, 5336.26it/s]                 
{&apos;class_name&apos;: &apos;Activation&apos;, &apos;config&apos;: {&apos;name&apos;: &apos;conv2d_2_tanh&apos;, &apos;trainable&apos;: True, &apos;dtype&apos;: &apos;float32&apos;, &apos;activation&apos;: &apos;tanh&apos;}}
Traceback (most recent call last):
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/bin/xnnc-run&quot;, line 33, in &lt;module&gt;
    sys.exit(load_entry_point(&apos;xnnc==2.0.0&apos;, &apos;console_scripts&apos;, &apos;xnnc-run&apos;)())
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/__main__.py&quot;, line 49, in main
    runner.normal_run(args)
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/runner.py&quot;, line 123, in normal_run
    target=target,
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/xconverter.py&quot;, line 145, in run
    model_files, model_type, _layout, in_shapes, batchsize
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/core.py&quot;, line 123, in make_xmodel
    model_type=model_t,
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py&quot;, line 107, in to_xmodel
    model_type,
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py&quot;, line 177, in create_xmodel
    name, layers, layout, in_shapes, batchsize
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py&quot;, line 2192, in __create_xmodel_from_tf2
    xnode.tmp_params[&quot;name&quot;] = layer[&apos;name&apos;]
KeyError: &apos;name&apos;
</pre>

Which got fixed similarly.
Afterwards, I got this error

<pre>[INFO] keras version: 2.6.0
[INFO] Tensorflow Keras model type: sequential
[INFO] parse raw model     :  0%|          | 0/18 [00:00&lt;?, ?it/s]              [INFO] parse raw model     :100%|██████████| 18/18 [00:00&lt;00:00, 12872.54it/s]               
[INFO] infer shape (NHWC)  :  0%|          | 0/31 [00:00&lt;?, ?it/s]              [INFO] infer shape (NHWC)  : 32%|███▏      | 10/31 [00:00&lt;00:00, 11798.32it/s]               
Traceback (most recent call last):
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/bin/xnnc-run&quot;, line 33, in &lt;module&gt;
    sys.exit(load_entry_point(&apos;xnnc==2.0.0&apos;, &apos;console_scripts&apos;, &apos;xnnc-run&apos;)())
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/__main__.py&quot;, line 49, in main
    runner.normal_run(args)
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/runner.py&quot;, line 123, in normal_run
    target=target,
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/xconverter.py&quot;, line 145, in run
    model_files, model_type, _layout, in_shapes, batchsize
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/core.py&quot;, line 123, in make_xmodel
    model_type=model_t,
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py&quot;, line 107, in to_xmodel
    model_type,
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py&quot;, line 177, in create_xmodel
    name, layers, layout, in_shapes, batchsize
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py&quot;, line 2310, in __create_xmodel_from_tf2
    if not xmodel.infer_shape(Layout.NHWC):
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/ir/xmodel.py&quot;, line 504, in infer_shape
    ok, error = self.__do_shape_inference(Layout[self.layout])
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/ir/xmodel.py&quot;, line 529, in __do_shape_inference
    xnode.infer_shape(layout)
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/ir/xnode.py&quot;, line 6926, in infer_shape
    shape = self.tmp_params[&apos;shape&apos;]
KeyError: &apos;shape&apos;</pre>

Printing the self.tmp_params, I got
<pre>{&apos;name&apos;: &apos;conv2d_2_tanh&apos;, &apos;trainable&apos;: True, &apos;data_type&apos;: &apos;float32&apos;, &apos;activation&apos;: &apos;tanh&apos;, &apos;kind&apos;: &apos;Activation&apos;}
</pre>

shape completely missing.

I am attaching my float model and my quant model
[Models.zip](https://github.com/Xilinx/Vitis-AI/files/8521705/Models.zip)

Also this is the code of my model

model = keras.Sequential([
    layers.Input(shape=32,32,1),
    layers.Conv2D(6, kernel_size=(5, 5), strides=(1, 1), activation='tanh', padding="valid"),
    layers.AveragePooling2D(pool_size=(2, 2), strides=(2, 2), padding='valid'),
    layers.Conv2D(16, kernel_size=(5, 5), strides=(1, 1), activation='tanh', padding='valid'),
    layers.AveragePooling2D(pool_size=(2, 2), strides=(2, 2), padding='valid'),
    layers.Flatten(),
    layers.Dense(units=120, activation='tanh'),
    layers.Dense(units=84, activation='tanh'),
    layers.Dense(units=10)
])

jimheaton, 2022-04-20:
I do not think sequential style models are supported. Try rewriting your model using keras functional APIs.

aimilefth, 2022-04-28:
Thanks for the response

I rewrited my model this way:

inputs = layers.Input(shape=INPUT_SHAPE)
x = layers.Conv2D(6, kernel_size=(5, 5), strides=(1, 1), activation='tanh', padding="valid")(inputs)
x = layers.AveragePooling2D(pool_size=(2, 2), strides=(2, 2), padding='valid')(x)
x = layers.Conv2D(16, kernel_size=(5, 5), strides=(1, 1), activation='tanh', padding='valid')(x)
x = layers.AveragePooling2D(pool_size=(2, 2), strides=(2, 2), padding='valid')(x)
x = layers.Flatten()(x)
x = layers.Dense(units=120, activation='tanh')(x)
x = layers.Dense(units=84, activation='tanh')(x)
outputs = layers.Dense(units=10)(x)

model = keras.Model(inputs=inputs, outputs=outputs, name="LeNet5")

The first 2 errors were indeed fixed with this change
However the last error (Keynote:shape) occurred again. I repost it 
<pre>COMPILING MODEL FOR ALVEO U50..
-----------------------------------------
[INFO] Namespace(batchsize=1, inputs_shape=None, layout=&apos;NHWC&apos;, model_files=[&apos;build/quant_model/q_model.h5&apos;], model_type=&apos;tensorflow2&apos;, named_inputs_shape=None, out_filename=&apos;/tmp/customcnn_org.xmodel&apos;, proto=None)
[INFO] tensorflow2 model: /workspace/tutorials/CNN_Model_Benchmarking/LeNet5/files/build/quant_model/q_model.h5
[INFO] keras version: 2.6.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :  0%|          | 0/18 [00:00&lt;?, ?it/s]              [INFO] parse raw model     :100%|██████████| 18/18 [00:00&lt;00:00, 9141.24it/s]                
[INFO] infer shape (NHWC)  :  0%|          | 0/31 [00:00&lt;?, ?it/s]              [INFO] infer shape (NHWC)  : 32%|███▏      | 10/31 [00:00&lt;00:00, 11580.08it/s]               
Traceback (most recent call last):
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/bin/xnnc-run&quot;, line 33, in &lt;module&gt;
    sys.exit(load_entry_point(&apos;xnnc==2.0.0&apos;, &apos;console_scripts&apos;, &apos;xnnc-run&apos;)())
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/__main__.py&quot;, line 49, in main
    runner.normal_run(args)
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/runner.py&quot;, line 123, in normal_run
    target=target,
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/xconverter.py&quot;, line 145, in run
    model_files, model_type, _layout, in_shapes, batchsize
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/core.py&quot;, line 123, in make_xmodel
    model_type=model_t,
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py&quot;, line 107, in to_xmodel
    model_type,
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py&quot;, line 177, in create_xmodel
    name, layers, layout, in_shapes, batchsize
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py&quot;, line 2310, in __create_xmodel_from_tf2
    if not xmodel.infer_shape(Layout.NHWC):
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/ir/xmodel.py&quot;, line 504, in infer_shape
    ok, error = self.__do_shape_inference(Layout[self.layout])
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/ir/xmodel.py&quot;, line 529, in __do_shape_inference
    xnode.infer_shape(layout)
  File &quot;/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/ir/xnode.py&quot;, line 6926, in infer_shape
    shape = self.tmp_params[&apos;shape&apos;]
KeyError: &apos;shape&apos;</pre>

I tried to get rid of the tanh and replaced it with relu and the issue didnt come up.
Hence I am positive that the issue is tanh specific. Still, it shouldn't happen and needs fixing, am I wrong?

zhenzhen-AMD, 2022-07-25:
Hi @aimilefth ，
We will give you a patch later to fix it. thanks.

zhenzhen-AMD, 2022-07-29:
Hi @aimilefth ,

The patch has been generated. Please use the following command to update:
wget -O xnnc-2.5.0-py36hd51751d_201.tar.bz2  https://www.xilinx.com/bin/public/openDownload?filename=xnnc-2.5.0-py36hd51751d_201.tar.bz2
wget -O xnnc-2.5.0-py37hd51751d_201.tar.bz2  https://www.xilinx.com/bin/public/openDownload?filename=xnnc-2.5.0-py37hd51751d_201.tar.bz2
inside docker. : 
conda update -n vitis-ai-tensorflow xnnc-2.5.0-py36hd51751d_201.tar.bz2
conda update -n vitis-ai-tensorflow2 xnnc-2.5.0-py37hd51751d_201.tar.bz2

renjiangyong, 2022-09-06:
The issue opened for more than one month, @aimilefth , I will close it , U can reopen it when you need.

closed 2022-09-06 by renjiangyong