Xilinx/Vitis-AI #766: Vitis AI 2.0 VCK190ES1Failed to open video: video/traffic.webm

AfifaIshtiaq, 2022-04-21:
root@versal-rootfs-common-2021_2:~/demo/VART/adas_detection# ./adas_detection video/adas.webm yolov3_adas_pruned_0_9.xmodel
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0421 11:59:30.345094  1148 main.cc:391] create running for subgraph: subgraph_layer0-conv
XAIEFAL: INFO: Resource group Avail is created.
XAIEFAL: INFO: Resource group Static is created.
XAIEFAL: INFO: Resource group Generic is created.
[ WARN:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ERROR:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap.cpp (140) open VIDEOIO(GSTREAMER): raised OpenCV exception:

OpenCV(4.4.0) /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_gstreamer.cpp:743: error: (-215:Assertion failed) uridecodebin in function 'open'

[ERROR:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap.cpp (140) open VIDEOIO(CV_IMAGES): raised OpenCV exception:

OpenCV(4.4.0) /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_images.cpp:253: error: (-5:Bad argument) CAP_IMAGES: can't find starting number (in the name of file): video/adas.webm in function 'icvExtractPattern'

Fail to open specified video file:video/adas.webm
root@versal-rootfs-common-2021_2:~/demo/VART/adas_detection# cd ..
root@versal-rootfs-common-2021_2:~/demo/VART# cd segmentation/
root@versal-rootfs-common-2021_2:~/demo/VART/segmentation# ls
build.sh  fpn.xmodel  readme  segmentation  src  video
root@versal-rootfs-common-2021_2:~/demo/VART/segmentation#  ./segmentation video/traffic.webm fpn.xmodel
Detect video: video/traffic.webm
[ WARN:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ERROR:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap.cpp (140) open VIDEOIO(GSTREAMER): raised OpenCV exception:

OpenCV(4.4.0) /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_gstreamer.cpp:743: error: (-215:Assertion failed) uridecodebin in function 'open'

[ERROR:0] global /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap.cpp (140) open VIDEOIO(CV_IMAGES): raised OpenCV exception:

OpenCV(4.4.0) /usr/src/debug/opencv/4.4.0-r0/git/modules/videoio/src/cap_images.cpp:253: error: (-5:Bad argument) CAP_IMAGES: can't find starting number (in the name of file): video/traffic.webm in function 'icvExtractPattern'

Failed to open video: video/traffic.webm

tianfang-fafafa, 2022-04-22:
hi @AfifaIshtiaq , where do get this image or how do generate it?  could you please share more details?

AfifaIshtiaq, 2022-04-22:
I generated it myself using the steps in vitas-ai/dsa/xvdpu-trd
This image works fine with examples using images I only face issue when examples want to use a video file that I downloaded from running vitis ai examples 
VitisAI/vart/

tianfang-fafafa, 2022-04-22:
@AfifaIshtiaq ， I think you use the steps in this file https://github.com/Xilinx/Vitis-AI/blob/master/dsa/XVDPU-TRD/vck190_platform/README.md

if right, your rootfs will not be common rootfs.

AfifaIshtiaq, 2022-04-22:
Then what it should be it boots with versal-rootfs-common?
How it doesn’t allow the video example to run while image examples run fine

tianfang-fafafa, 2022-04-22:
@AfifaIshtiaq  in this file https://github.com/Xilinx/Vitis-AI/blob/master/dsa/XVDPU-TRD/vck190_platform/petalinux/xilinx-vck190-base-trd/project-spec/configs/config :
         CONFIG_SUBSYSTEM_HOSTNAME="xilinx-vck190-2021_2"
I think you should see "root@xilinx-vck190-2021_2" in your system ternimal.
you may use incorrected rootfs.tar.gz while generate sdcard.img which doesn't include some packags for video

AfifaIshtiaq, 2022-04-22:
what rootfs should be used?

tianfang-fafafa, 2022-04-22:
@AfifaIshtiaq  could you please share your step for details?

AfifaIshtiaq, 2022-04-22:
I ran the steps from https://github.com/Xilinx/Vitis-AI/tree/master/dsa/XVDPU-TRD but replaced the platform with the ES1 default from https://www.xilinx.com/member/forms/download/design-license-vck190-es1-base-20212-xef.html?filename=xilinx_vck190_es1_base_202120_1.zip

as described in 6.2 of the first link

tianfang-fafafa, 2022-04-22:
@AfifaIshtiaq 
 I see.  you change platform for vck es1 board.  how do you set  --package.rootfs ?
you should set  --package.rootfs with rootfs.ext4 which is generate by https://github.com/Xilinx/Vitis-AI/tree/master/dsa/XVDPU-TRD/vck190_platform/petalinux/xilinx-vck190-base-trd/

after generating, rootfs.ext4 will be https://github.com/Xilinx/Vitis-AI/tree/master/dsa/XVDPU-TRD/vck190_platform/petalinux/xilinx-vck190-base-trd/images/linux/

AfifaIshtiaq, 2022-04-22:
Thanks I'll try it

closed 2022-04-26 by AfifaIshtiaq