Xilinx/Vitis-AI #768: Tensorflow Quantization error - ModuleNotFoundError: No module named 'tensorflow.contrib.decent_q'; 'tensorflow.contrib' is not a package

xelalin, 2022-04-22:
When I done to build up GPU/CPU Docker from recipe, I launch vai_q_tensorflow to do quantization and found below error message on Anaconda vitis-ai-tensorflow environment. how to fix the problem?

> Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/vai_q_tensorflow", line 7, in <module>
    from tensorflow.contrib.decent_q.python.decent_q import run_main
ModuleNotFoundError: No module named 'tensorflow.contrib.decent_q'; 'tensorflow.contrib' is not a package

renjiangyong, 2022-07-22:
U can try the lastest Vitis-AI  2.5 docker,   https://hub.docker.com/layers/vitis-ai/xilinx/vitis-ai/2.5/images/sha256-eaa85efb06924995ebdb973546e7f69169b003b8cc525764bd9524ad554dddbe?context=explore. Thank you

renjiangyong, 2022-09-06:
The issue opened for more than a month, I will close it , you can reopen it when you need.

closed 2022-09-06 by renjiangyong

haimzisman, 2023-06-27:
Have this error too, and I need to use Vitis-AI-3.0.

amonni5, 2024-03-21:
Same issue, Vitis-AI 3.5 gpu mode