Xilinx/Vitis-AI #770: Segmentation fault during DPU Build in Vitis IDE

Hamptonjc, 2022-04-25:
I am attempting to build a 3-core B4096 DPU via the Vitis flow using the Vitis IDE. However I get a segmentation error once synthesis starts. I have traced down the issue to Vivado not having enough stack memory (https://support.xilinx.com/s/article/64434?language=en_US).

**Is there a way to adjust Vivado's stack size for v++ (possibly via the prj_config file)?** 

I believe I've found a temporary workaround, by setting the vivado setting synth.jobs=1 & impl.jobs=1 in the prj_config file. However this will cause the build to be painfully slow.

jiaz-xlnx, 2022-04-26:
Hi @Hamptonjc 
I think you have got the right way to add the property in the prj_config file. You can adjust the jobs number based on you server hardware. maybe try jobs=2 or 3. It will use less time.

Hamptonjc, 2022-04-26:
@jiaz-xlnx With synth.jobs=1 & impl.jobs=1, I get passed synthesis. However it still fails with the same error during implementation.

jiaz-xlnx, 2022-04-27:
Hi @Hamptonjc  Can you post your server details? Vitis has minimum computer requirements. You can check this link. [https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/Installation-Requirements](url)

Hamptonjc, 2022-04-27:
@jiaz-xlnx My host seems to meet requirements:

OS: Ubuntu 20.04.3 LTS

Memory: 32G

CPU: Intel 12-core i5

jiaz-xlnx, 2022-04-27:
Hi @Hamptonjc  Can you try intergrate 1 dpu core in the platform?

Hamptonjc, 2022-04-27:
@jiaz-xlnx I have been successful building a B4096 2-core DPU. Seems when I go to 3-cores, this issue happens.