Xilinx/Vitis-AI #771: csh or bash shell?

giovanniguasti, 2022-04-26:
https://github.com/Xilinx/Vitis-AI/tree/master/dsa/XVDPU-TRD/README.md

in the same .md instruction page we first require the usage of **csh**, then to set the terminal to **bash** shell

xlinyg, 2022-05-05:
Hi, @giovanniguasti 
Scripts for compiling DPU's AIE are designed as "*.csh", so **csh** is required.
But just need installing **csh** on your server, it will be called automatically.
You just need to run the XVDPU-TRD in the bash shell.

qianglin-xlnx, 2022-07-27:
Hi @giovanniguasti 
Did @xlinyg 's answer address your question? or do you have other questions?
https://github.com/Xilinx/Vitis-AI/issues/771#issuecomment-1118103562

giovanniguasti, 2022-07-27:
hi @qianglin-xlnx  and @xlinyg, 
thanks for the answer, the question is answered

closed 2022-07-27 by giovanniguasti