Xilinx/Vitis-AI #773: Does vitis-AI1.4 the official image for zcu102 support direct connection to DisplayPort to achieve image and video classification and detection results? How can I display the results?

LoeWHJ, 2022-04-26:


zz002, 2022-04-27:
hi @LoeWHJ 
You can use the USB micro-AB connector to connect a mouse and keyboard to the ZCU102, and use DisplayPort to connect a monitor.
Write a program like [demo](https://github.com/Xilinx/Vitis-AI/blob/1.4/tools/Vitis-AI-Runtime/VART/vart/dpu-runner/samples/adas_detection/src/main.cc#L178), and then run it in the system of zcu102, instead of running it through remote login
       cv::imshow("ADAS Detection@Xilinx DPU", frame);

qianglin-xlnx, 2022-06-23:
Hi @LoeWHJ 
Has this issue been solved?

qianglin-xlnx, 2022-06-28:
Hi @LoeWHJ 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-06-28 by qianglin-xlnx