Xilinx/Vitis-AI #774: Weird Subgraph for Flatten Layer

HenningCode, 2022-04-26:
Hello,

I am trying to run some small Models on the DPU and I wanted to look at the Subgraphs, but the Flatten Layers seems to look a bit weird. 
This is the Subgraph with `xdputil` 
[image]

and this with Netron: 
[image]
 
This Model works, but I do want to understand why the Subgraph in Netron adds another Conv2d Layer and is reshaping twice?

Greetings

wanghong4compiler, 2022-05-24:
Hi @HenningCode ,
Dense layers would be converted to Matmul layers, Matmul layers would be converted to Conv2d layers because it's a conv like operation.
The first one is added when transfer Matmul to Conv2d in order to fit hardware, and the second one is added to recover the dimension of the tensor for softmax.
The added two reshape operators do not drop performance, because they don't generate instructions actually.

qianglin-xlnx, 2022-07-13:
Hi @HenningCode 
Is this issue solved?

qianglin-xlnx, 2022-07-18:
Hi @HenningCode 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-07-18 by qianglin-xlnx