Xilinx/Vitis-AI #775: Batch Size Not Set During PyTorch Fast Finetune of Fully-Connected Network

Hamptonjc, 2022-04-26:
Model:
```python 
net = torch.nn.Sequential(torch.nn.Linear(128, 64), torch.nn.ReLU(),
                                    torch.nn.Linear(64, 32), torch.nn.ReLU(),
                                    torch.nn.Linear(32, 16), torch.nn.ReLU(),
                                    torch.nn.Linear(16, 2), torch.nn.Sigmoid())
```

During quantization (with fast finetune) I get:
```
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 94, in fast_finetune 
self.processor.finetune(run_fn, run_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 164, in finetune 
self.adaquant.finetune_v2(run_fn, run_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/adaquant.py", line 763, in finetune_v2 
is_cached = self.is_cached(qnode, len(net_inputs[0]))
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/adaquant.py", line 780, in is_cached 
input_numel = np.prod([self._batch_size] + self.graph.node(node.in_nodes[0]).out_tensors[0].shape[1:])
  File "<__array_function__ internals>", line 6, in prod
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/numpy/core/fromnumeric.py", line 2911, in prod
    keepdims=keepdims, initial=initial, where=where)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/numpy/core/fromnumeric.py", line 90, in _wrapreduction
    return ufunc.reduce(obj, axis, dtype, out, **passkwargs)
TypeError: unsupported operand type(s) for *: 'NoneType' and 'int' 
```

I traced down the issue to the hook in fast finetune that sets the batch size: [link](https://github.com/Xilinx/Vitis-AI/blob/996459fb96cb077ed2f7e789d515893b1cccbc95/tools/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/qproc/adaquant.py#L229)
```python
  def hook_batch_size(self, hook_mods):
    def hook(module, input, output):
      if isinstance(output, torch.Tensor) and output.ndim == 4:
        self._batch_size = output.size()[0]
```
seems that self._batch_size is only set with output with 4 dimensions. The input dimensions of my fully-connected network is (batch_size, vector_length) thus only 2 dimensions throughout the network.

Any solution for this? Any work around?

Hamptonjc, 2022-04-27:
A work around for me is to change:
```python
  def hook_batch_size(self, hook_mods):
    def hook(module, input, output):
      if isinstance(output, torch.Tensor) and output.ndim == 4:
        self._batch_size = output.size()[0]
```
to:
```python
  def hook_batch_size(self, hook_mods):
    def hook(module, input, output):
      if isinstance(output, torch.Tensor) and (output.ndim == 4 or output.ndim == 2):
        self._batch_size = output.size()[0]
```

Maybe a better way to get batch size should be implemented here?

zl200881, 2022-05-23:
The Linear network is not supported by fast finetune since model quantization and fast finetune only support networks like CNN.
Using fast finetune may cause problems like "Batch Size Not Set". 
Fast finetune does not support the Linear net. It only  supports network like CNNs.

closed 2022-06-13 by niuxjxlnx