Xilinx/Vitis-AI #776: Vitis-AI quantizer

AfifaIshtiaq, 2022-04-28:
If we are running our own model with data set of our own do we have to write pre-processing script input_fn on our own? Is the pre-processing script same for all the frameworks like pytorch ,caffee and tensorflow? Do we  resize all the inputs to 224x224x3 before giving to pre-processing script or it is done by pre-processing script itself

AfifaIshtiaq, 2022-05-05:
I wrote the preprossing script and I had the following error while quantizing 
[image]

xieyi4650, 2022-07-21:
Hi @AfifaIshtiaq 
1. pre-processing is the same as float model pre-processing when running inference.
2. can you upload full error log?

renjiangyong, 2022-09-06:
Hi @AfifaIshtiaq , Does issue resolved?  the issue is existed for a long time , if you still have problem, you can reopen it .

closed 2022-09-06 by renjiangyong