Xilinx/Vitis-AI #777: About how to divide a .xmodel

jx991220, 2022-04-29:
I found that the total execution time of each layer in vaitrace "fine_grained" mode is greater than that of the network that does not use "fine_grained".

May I ask why? Or is there a way to read the .xmodel file and divide a network into several networks according to the order of DPU execution and execute them separately? (because I know that many layers are not parallel, so it doesn't affect the total execution time)?

li900309, 2022-06-15:
hi @jx991220 , while fine grained profiling model is enabled. dnn model will be spitted into several sub-graphs(one layer -> one sub-graph) and executed one-by-one.
1. It will introduce significant scheduling and memory access overhead, especially for small models.
2. Some xcompiler's optimizations (like layer fusion) will be turned-off.

li900309, 2022-06-15:
and I don't think we can dictatorially edit the xmodel . you can modify the original pytorch/tf model then re do the vitis-ai quantization and compile flow to get a new .xmodel

qianglin-xlnx, 2022-07-13:
Hi @jx991220 
Is this issue solved?

qianglin-xlnx, 2022-07-18:
Hi @jx991220 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-07-18 by qianglin-xlnx