Xilinx/Vitis-AI #778: Different output between U50 and U250

estherbernard, 2022-05-01:
Hi,

I made a decoder model from pytorch with Transpose2d and convolutions layers and this one works fine on the U50 with an output of size 1088x1920x3. However, while trying to deploy it on the U250, the results are really bad (and still 1088x1920x3), as if there was a problem while executing it on the U250 (it is not just about bad results, but instead a problem on the output).

By the way, while compiling the model for the u250, i arrive on the pdb deboguer, if i type the c command (or "continue") on the debugger, the compilation seems to work fine. But regardless to the results, maybe the problems come from this compilation:

Step  decoder_cond__decoder_cond_ConvTranspose2d_deconv2__input
WARNING: this tensor cannot be allocated with a stride Name decoder_cond__decoder_cond_Conv2d_conv1__305-decoder_cond__decoder_cond_ConvTranspose2d_deconv2__input Parameter False	Space 50135040 bits, BitePerPixel 8 Fraction 8 Start 0 End 6266880	Specifier  Layout -1 Time 0 Strategies None None	Shape  [1, 136, 240, 192] CNN_Shape TensorShapes(batch=1, height=271, width=479, channel=192) 1
Memory access IN ddr PAR pa  TMP ddr OUT ddr decoder_cond__decoder_cond_ConvTranspose2d_deconv2__input 
> /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/SC/HwAbstraction/code_generation.py(303)__init__()
-> self.code_engine = U_C(c,hw)
(Pdb) Step  decoder_cond__decoder_cond_Conv2d_conv2__343
Memory access IN ddr PAR pa  TMP fm  OUT ddr decoder_cond__decoder_cond_Conv2d_conv2__343 
Step  decoder_cond__decoder_cond_ConvTranspose2d_deconv3__362-conv-channel
Memory access IN ddr PAR pa  TMP ddr OUT ddr decoder_cond__decoder_cond_ConvTranspose2d_deconv3__362-conv-channel 
 ############################################# 
 ######  Naive instruction dependency
 ############################################# 
 ############################################# 
 ######  Code Generation at Node Level and then Recursively
 ############################################# 

Is there a problem on the second ConvTranspose2d?

I put the U50 and U250 compilation logs.

Thanks in advance.

[compile_output_u50.txt](https://github.com/Xilinx/Vitis-AI/files/8599096/compile_output_u50.txt)
[compile_output_u250.txt](https://github.com/Xilinx/Vitis-AI/files/8599098/compile_output_u250.txt)

paolodalberto, 2022-07-28:
This is a very good example. thank you.
Is it possible for you to share the model ?

The fact that there was a debug trace, it means the compiler was not happy and unlikely the code generation reached completion. The Warning could be harmless if the process split the computation by "width" ... but this is a transpose convolution. it could be tricky.

paolodalberto, 2022-08-13:
I have found that, as today, we do not support a transpose that requires a upsample larger that 2 when we deploy the optimized transposed (as a set of convolutions). However, the usual upsample-convolution is supported ... 

@estherbernard  it is the second transpose the problem. 
please share the model if possible ...