Xilinx/Vitis-AI #779: A task quantification for Youtube-8M

jia0511, 2022-05-02:
Dear xilinx,
I considered deploying on xilinx devices by quantifying the Youtube-8M classification problem through the pytorch quantization tool of vitis-ai (https://github.com/ceshine/yt8m-2019), but the quantization process encountered a series of problems, I tried I used different quantization tools such as CV/LSTM/RNN, but they all encountered various problems. Have you considered the quantitative deployment of video understanding tasks? Can you help us to see what modifications need to be made if such tasks need to be deployed.
-----------------------------------------------------------------------------
###########################CV-CALIB-ERROR#############################
(vitis-ai-pytorch) Vitis-AI /workspace/videounderstanding/pytorch-yt8m-infer > sh run_infer.sh 

[VAIQ_NOTE]: Loading NNDCT kernels...
batch_size 1

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cuda'.

[VAIQ_NOTE]: =>Parsing SampleFrameModelWrapper...

[VAIQ_NOTE]: Start to trace model...
/workspace/videounderstanding/pytorch-yt8m-infer/models.py:27: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  if features.size(1) > self.max_len:
/workspace/videounderstanding/pytorch-yt8m-infer/models.py:29: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  np.arange(features.size(1)),
/workspace/videounderstanding/pytorch-yt8m-infer/models.py:29: TracerWarning: Converting a tensor to a Python float might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  np.arange(features.size(1)),

[VAIQ_NOTE]: Finish tracing.
Traceback (most recent call last):
  File "run_infer.py", line 250, in <module>
    main()
  File "run_infer.py", line 233, in main
    quantizer = torch_quantizer(args.quant_mode, model, (input), device=device)   
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 89, in __init__
    custom_quant_ops = custom_quant_ops)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 126, in __init__
    device=device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 196, in prepare_quantizable_module
    graph = parse_module(module, input_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 78, in parse_module
    module, input_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/parser.py", line 68, in __call__
    raw_graph, raw_params = graph_handler.build_torch_graph(graph_name, module, input_args) 
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/trace_helper.py", line 48, in build_torch_graph
    opt_graph = self._opt_raw_graph(raw_graph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/trace_helper.py", line 113, in _opt_raw_graph
    OptPass.merge_matmul_with_add(raw_graph, node_sets)                                                
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/opt_pass.py", line 332, in merge_matmul_with_add
    raw_graph.reconnect_nodes(raw_graph)
TypeError: reconnect_nodes() takes 1 positional argument but 2 were given

-----------------------------------------------------------------------------
#######CV-TEST-ERROR#################

(vitis-ai-pytorch) Vitis-AI /workspace/videounderstanding/pytorch-yt8m-infer > sh run_infer.sh 

[VAIQ_NOTE]: Loading NNDCT kernels...
batch_size 1

[VAIQ_NOTE]: Quantization test process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cuda'.

[VAIQ_NOTE]: =>Parsing SampleFrameModelWrapper...

[VAIQ_NOTE]: Start to trace model...
/workspace/videounderstanding/pytorch-yt8m-infer/models.py:27: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  if features.size(1) > self.max_len:
/workspace/videounderstanding/pytorch-yt8m-infer/models.py:29: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  np.arange(features.size(1)),
/workspace/videounderstanding/pytorch-yt8m-infer/models.py:29: TracerWarning: Converting a tensor to a Python float might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  np.arange(features.size(1)),

[VAIQ_NOTE]: Finish tracing.

[VAIQ_NOTE]: Processing ops...
██████████████████████████████████████████████████| 42/42 [00:00<00:00, 2937.19it/s, OpInfo: name = SampleFrameModelWrapper/GatedDBoFModel[model]/1002, type = sum]                  

[VAIQ_WARN]: The quantizer recognize new op `index` as a float operator by default.

[VAIQ_WARN]: The quantizer recognize new op `aten::adaptive_max_pool1d` as a float operator by default.

[VAIQ_WARN]: The quantizer recognize new op `aten::adaptive_avg_pool1d` as a float operator by default.

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(quantize_result/SampleFrameModelWrapper.py)

[VAIQ_NOTE]: =>Converting to xmodel ...
Traceback (most recent call last):
  File "run_infer.py", line 250, in <module>
    main()
  File "run_infer.py", line 239, in main
    quantizer.export_xmodel(deploy_check=False)      
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 110, in export_xmodel
    self.processor.export_xmodel(output_dir, deploy_check)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 180, in export_xmodel
    dump_xmodel(output_dir, deploy_check, self._lstm_app)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 203, in dump_xmodel
    deploy_graphs = get_deploy_graph_list(quantizer.quant_model, quantizer.Nndctgraph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 290, in get_deploy_graph_list
    update_nndct_blob_data(quant_model, g_optmizer.frozen_graph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 165, in update_nndct_blob_data
    dfs(source, visited)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 156, in dfs
    dfs(cn, visited)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 156, in dfs
    dfs(cn, visited)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 156, in dfs
    dfs(cn, visited)
  [Previous line repeated 9 more times]
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 152, in dfs
    node.out_tensors[0].from_ndarray(data)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_tensor.py", line 66, in from_ndarray
    raise TypeError("'data' must be a numpy ndarray")
TypeError: 'data' must be a numpy ndarray
----------------------------------------------------------------------------------------------------------------------------------

niuxjxlnx, 2022-05-30:
hi, @jia0511:

Supposing you are using Vitis-AI 2.0, you can change this line 
/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/opt_pass.py", line 332
from 
`raw_graph.reconnect_nodes(raw_graph)`
to 
`raw_graph.reconnect_nodes()`
to bypass the bug.

jia0511, 2022-05-30:
@niuxjxlnx 

Thanks for answering this question！Yes, I changed this line from
raw_graph.reconnect_nodes(raw_graph)
to
raw_graph.reconnect_nodes()
but now have the new question, you can see the "CV-TEST-ERROR".

niuxjxlnx, 2022-05-31:
@jia0511 :

I mean the final error in part "CV-TEST-ERROR":
`"TypeError: 'data' must be a numpy ndarray",` 
is caused by that no model inference is called before you call the function of converting xmodel. 

On how to do model inference before converting xmodel, please refer to example code in [resnet18_quant.py](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Quantizer/vai_q_pytorch/example/resnet18_quant.py#L249).

jia0511, 2022-08-03:
Hi @zl200881 ,

We have modified the sigmod operator to be a Relu operator, AvgMaxPooler is a class, not a real operator, defined as follows, the input dimension is a fixed input <1,300,1152> during training, but it still exists in the end
"TypeError: reconnect_nodes() takes 1 positional argument but 2 were given" problem.
[image]

niuxjxlnx, 2022-08-31:
@jia0511 :

Did you do the modification?

[/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/parse/opt_pass.py", line 332
from
raw_graph.reconnect_nodes(raw_graph)
to
raw_graph.reconnect_nodes()]

jia0511, 2022-08-31:
@niuxjxlnx 

Yes, I have tried it, so I can run the run_infer.sh script, but there are still problems running.
Please check the #######CV-TEST-ERROR#################  section for errors.

zl200881, 2022-09-01:
Hi @jia0511, I don't encounter your errors. I quantize the model with the attached code and get the result successfully.
You can try it by unzipping the code in your directory and run:
_sh run_quant.sh_
[run.zip](https://github.com/Xilinx/Vitis-AI/files/9469624/run.zip)

jia0511, 2022-09-02:
@zl200881 

Thank you for your support. At present, I can quantify normally in vitis-ai2.5 version, but some operators are still running on the CPU, and there is still a certain decline in performance. In addition, can you provide reference C++ code for hardware deployment, so as to facilitate We deploy further, measure FPS and accuracy？

zl200881, 2022-09-05:
@jia0511 :
1. The C++ code for deployment is in [xir](https://github.com/Xilinx/Vitis-AI/tree/c26eae36f034d5a2f9b2a7bfe816b8c43311a4f8/src/Vitis-AI-Runtime/VART/xir)
2. I have checked through the fix-point in quantize_result/quant_info.json and find it reaches as large as -26 which may result in performance decline. Please check your model for better performance.

niuxjxlnx, 2022-09-05:
Close the issue since quantization issues are solved. Go ahead to open other issues for more problems needs to be solved.

closed 2022-09-05 by niuxjxlnx