Xilinx/Vitis-AI #782: docker_build_gpu fails because of NVIDA pub key

ydajwz123, 2022-05-04:
I just share method to continue build for someone encountering this problem.
Some command could be added to GPU Dockerfile. Such as 
```
apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/3bf863cc.pub
```

jzymessi, 2022-05-09:
I added the above command to the vitis-ai-gpu.Dockerfile file according to your method, and the following problem occurred. How to solve it, thank you.
[image]

ydajwz123, 2022-05-09:
> I added the above command to the vitis-ai-gpu.Dockerfile file according to your method, and the following problem occurred. How to solve it, thank you. [image]
Do you build GPU docker successfully? I did some changes such as following lines.
```shell
  29 RUN apt-key adv --fetch-keys https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/3bf863cc.pub \
  30     && sed -i s/archive.ubuntu.com/mirrors.aliyun.com/g /etc/apt/sources.list \
  31     && sed -i s/security.ubuntu.com/mirrors.aliyun.com/g /etc/apt/sources.list \
```
However, it still doesn't work. You may find new issue in #739 .

janifer112x, 2022-09-05:
Hi @ydajwz123,
as we have release Vitis AI 2.5 this June, Could you please try with newer version?

janifer112x, 2022-09-06:
closed it with fix in master branch

closed 2022-09-06 by janifer112x