Xilinx/Vitis-AI #789: When will this library be updated to work with Vitis 2022.1?

nlbutts, 2022-05-06:
Trying to re-build the DPU for the ZU7 fails due to the gen_dpu_xo.tcl file requiring 4 parameters instead of 3 parameters. 

```
ERROR: Program "/home/nlbutts/projects/xilinx/Vitis-AI/dsa/DPU-TRD/prj/Vitis/scripts/gen_dpu_xo.tcl" requires 4 arguments!

Usage: /home/nlbutts/projects/xilinx/Vitis-AI/dsa/DPU-TRD/prj/Vitis/scripts/gen_dpu_xo.tcl <xoname> <krnl_name> <target> <device>
```

I tried to add a fourth parameter, but then I get the following error:
```/home/nlbutts/xilinx/Vivado/2022.1/bin/vivado -mode batch -source /home/nlbutts/projects/xilinx/Vitis-AI/dsa/DPU-TRD/prj/Vitis/scripts/gen_dpu_xo.tcl -notrace -tclargs binary_container_1/dpu.xo dpu.xo DPUCZDX8G hw 

****** Vivado v2022.1 (64-bit)
  **** SW Build 3526262 on Mon Apr 18 15:47:01 MDT 2022
  **** IP Build 3524634 on Mon Apr 18 20:55:01 MDT 2022
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

source /home/nlbutts/projects/xilinx/Vitis-AI/dsa/DPU-TRD/prj/Vitis/scripts/gen_dpu_xo.tcl -notrace
binary_container_1/dpu.xo
CRITICAL WARNING: [filemgmt 20-730] Could not find a top module in the fileset sources_1.
Resolution: With the gui up, review the source files in the Sources window. Use Add Sources to add any needed sources. If the files are disabled, enable them. You can also select the file and choose Set Used In from the pop-up menu. Review if they are being used at the proper points of the flow.
CRITICAL WARNING: [filemgmt 20-730] Could not find a top module in the fileset sim_1.
Resolution: With the gui up, review the source files in the Sources window. Use Add Sources to add any needed sources. If the files are disabled, enable them. You can also select the file and choose Set Used In from the pop-up menu. Review if they are being used at the proper points of the flow.
CRITICAL WARNING: [filemgmt 20-730] Could not find a top module in the fileset sources_1.
Resolution: With the gui up, review the source files in the Sources window. Use Add Sources to add any needed sources. If the files are disabled, enable them. You can also select the file and choose Set Used In from the pop-up menu. Review if they are being used at the proper points of the flow.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/nlbutts/projects/xilinx/Vitis-AI/dsa/DPU-TRD/dpu_ip/Vitis/dpu/hdl/DPUCZDX8G.v'.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/nlbutts/projects/xilinx/Vitis-AI/dsa/DPU-TRD/dpu_ip/Vitis/dpu/inc/arch_def.vh'.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/nlbutts/projects/xilinx/Vitis-AI/dsa/DPU-TRD/dpu_ip/DPUCZDX8G_v3_4_0/hdl/DPUCZDX8G_v3_4_0_vl_dpu.sv'.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/nlbutts/projects/xilinx/Vitis-AI/dsa/DPU-TRD/dpu_ip/DPUCZDX8G_v3_4_0/inc/arch_para.vh'.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/nlbutts/projects/xilinx/Vitis-AI/dsa/DPU-TRD/dpu_ip/DPUCZDX8G_v3_4_0/inc/function.vh'.
WARNING: [IP_Flow 19-3833] Unreferenced file from the top module is not packaged: '/home/nlbutts/projects/xilinx/Vitis-AI/dsa/DPU-TRD/prj/Vitis/dpu_conf.vh'.

    while executing
"ipx::package_project -root_dir $path_to_packaged -vendor xilinx.com -library RTLKernel -taxonomy /KernelIP -import_files -set_current false"
    (file "./scripts/package_dpu_kernel.tcl" line 37)

    while executing
"source -notrace ./scripts/package_dpu_kernel.tcl"
    invoked from within
"if { [info exists ::env(DIR_PATH)] } {
    source -notrace $env(DIR_PRJ)/scripts/package_dpu_kernel.tcl
} else {
    source -notrace ./scripts/package..."
    (file "/home/nlbutts/projects/xilinx/Vitis-AI/dsa/DPU-TRD/prj/Vitis/scripts/gen_dpu_xo.tcl" line 30)
INFO: [Common 17-206] Exiting Vivado at Fri May  6 11:48:50 2022...
make: *** [Makefile:70: binary_container_1/dpu.xo] Error 1
```

I just installed Vitis 2022.1 and pulled the latest version of this repo.
I sourced the Vitis environment, then defined
`EDGE_COMMON_SW`
and
`SDX_PLATFORM`

Using the 2022.1 ZCU-104 platform files. 

I then ran `make` in this directory `Vitis-AI/dsa/DPU-TRD/prj/Vitis`

tianfang-fafafa, 2022-05-13:
hi @nlbutts ,  DPU_TRD v2022.1 hasn't been released.  The directory you used doesn't match vivado v2022.1

tianfang-fafafa, 2022-07-14:
hi @nlbutts， has the problem been solved?

qianglin-xlnx, 2022-07-18:
Hi @nlbutts 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-07-18 by qianglin-xlnx