Xilinx/Vitis-AI #791: Model forward loop for evaluation is needed before exporting quantization config.

Gang-Chen-China, 2022-05-09:
Model forward loop for evaluation is needed before exporting quantization config.
Please refer to the function "evaluate" in the example resnet18_quant.py.

_Originally posted by @zl200881 in https://github.com/Xilinx/Vitis-AI/issues/761#issuecomment-1108046369_

niuxjxlnx, 2022-05-18:
@Gang-Chen-China :

This description looks like paste some replying contents from another issue but no issue is showed.
Please feel free to open another issue if you meet issue when you using Vitis-AI.

closed 2022-05-18 by niuxjxlnx

BusyBeeATais, 2023-08-28:
Why is no "evaluative" function needed for the MINST, but it is needed here? What is the difference between the two examples and when is this function needed?