Xilinx/Vitis-AI #792: Quantization vai_q

AfifaIshtiaq, 2022-05-09:


AfifaIshtiaq, 2022-05-09:
how can we freeze the graph in tensorflow 2.x? whats the cause of following issue?
(vitis-ai-tensorflow2) Vitis-AI /workspace/models/ChEstModel > python3 test_model.py 
2022-05-06 07:59:45.184408: I tensorflow/core/util/util.cc:159] Experimental oneDNN custom operations are on. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2022-05-06 07:59:45.188701: W tensorflow/stream_executor/platform/default/dso_loader.cc:64] Could not load dynamic library 'libcudart.so.11.0'; dlerror: libcudart.so.11.0: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: /opt/xilinx/xrt/lib:/usr/lib:/usr/lib/x86_64-linux-gnu:/usr/local/lib:/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib
2022-05-06 07:59:45.188715: I tensorflow/stream_executor/cuda/cudart_stub.cc:29] Ignore above cudart dlerror if you do not have a GPU set up on your machine.
2022-05-06 07:59:46.943853: W tensorflow/stream_executor/platform/default/dso_loader.cc:64] Could not load dynamic library 'libcuda.so.1'; dlerror: libcuda.so.1: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: /opt/xilinx/xrt/lib:/usr/lib:/usr/lib/x86_64-linux-gnu:/usr/local/lib:/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib
2022-05-06 07:59:46.943872: W tensorflow/stream_executor/cuda/cuda_driver.cc:269] failed call to cuInit: UNKNOWN ERROR (303)
2022-05-06 07:59:46.943889: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:156] kernel driver does not appear to be running on this host (dre-elbe-s04): /proc/driver/nvidia/version does not exist
2022-05-06 07:59:46.944150: I tensorflow/core/platform/cpu_feature_guard.cc:151] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
[VAI INFO] Update custom_layer_type: []
Traceback (most recent call last):
  File "test_model.py", line 47, in <module>
    quantized_model = quantizer.quantize_model(calib_dataset=test_generator)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 566, in quantize_model
    self.optimize_model()
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 454, in optimize_model
    self._create_optimized_model()
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 228, in _create_optimized_model
    quantize_strategy=self._quantize_strategy)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 700, in create_optimize_model
    model, candidate_layers, layer_metadata)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py", line 74, in apply
    model, configs, available_transforms, candidate_layers, layer_metadata)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py", line 50, in _apply_availables
    layer_metadata).recursive_transform()
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 738, in recursive_transform
    self.layer_metadata).transform()
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 704, in transform
    self._set_layer_weights(layer, weights_map)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 593, in _set_layer_weights
    K.batch_set_value(weight_value_tuples)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow/python/util/traceback_utils.py", line 153, in error_handler
    raise e.with_traceback(filtered_tb) from None
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/keras/backend.py", line 4019, in batch_set_value
    x.assign(np.asarray(value, dtype=dtype_numpy(x)))
ValueError: Cannot assign value to variable ' BN-2-0/gamma:0': Shape mismatch.The variable shape (64,), and the assigned value shape (128,) are incompatible.

foolmarks, 2022-05-09:
VitisAI for TF2 does not use frozen graphs,  see UG1414.Your error comes feom TF, not VitisAI. You have a mismatch between  sizes..perhaps you are assigning 128 to a 64 model input?Sent from my Galaxy

AfifaIshtiaq, 2022-05-09:
I use the model in this paper Fig 3 page 3929. It gives me error on layer 2 when we try to change dimension from 64 to 128

foolmarks, 2022-05-09:
Which paper?  Post a link.

AfifaIshtiaq, 2022-05-09:
> Which paper? Post a link.

[Uploading Honkala_Nokia_DeepRxFullyConvDeepLearningRx_IEEE_TWC_2021-06.pdf…]()

AfifaIshtiaq, 2022-05-09:
> > Which paper? Post a link.
> 
> Uploading Honkala_Nokia_DeepRxFullyConvDeepLearningRx_IEEE_TWC_2021-06.pdf…

https://ieeexplore.ieee.org/abstract/document/9345504

AfifaIshtiaq, 2022-05-09:
> > > Which paper? Post a link.
> > 
> > 
> > Uploading Honkala_Nokia_DeepRxFullyConvDeepLearningRx_IEEE_TWC_2021-06.pdf…
> 
> https://ieeexplore.ieee.org/abstract/document/9345504

[image]

foolmarks, 2022-05-09:
So you have a model with one one input or are doing the concatenation inside your model?

What do you mean by 'change dimension from 64 to 128'?   Where did you make this change?

AfifaIshtiaq, 2022-05-09:
> So you have a model with one one input or are doing the concatenation inside your model?
> 
> What do you mean by 'change dimension from 64 to 128'? Where did you make this change?

This is our input format  and model aslo expalined in the figure attached below. First I thought that the issue arise when we have upsampling in dimensions i.e., Resnet Block 1 (14,72,64 ) to Resnet Block 2 (14,72,128 ). But then I  generated model using only Resnet 0 block and Resnet 1 block with same dimensions and I observed the following:
Traceback (most recent call last):
  File "test_model.py", line 49, in <module>
    quantized_model = quantizer.quantize_model(calib_dataset=test_generator)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 566, in quantize_model
    self.optimize_model()
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 454, in optimize_model
    self._create_optimized_model()
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 228, in _create_optimized_model
    quantize_strategy=self._quantize_strategy)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 700, in create_optimize_model
    model, candidate_layers, layer_metadata)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py", line 92, in apply
    layer_metadata)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_transforms_pipeline.py", line 50, in _apply_availables
    layer_metadata).recursive_transform()
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 738, in recursive_transform
    self.layer_metadata).transform()
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 704, in transform
    self._set_layer_weights(layer, weights_map)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 593, in _set_layer_weights
    K.batch_set_value(weight_value_tuples)
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/tensorflow/python/util/traceback_utils.py", line 153, in error_handler
    raise e.with_traceback(filtered_tb) from None
  File "/home/vitis-ai-user/.local/lib/python3.7/site-packages/keras/backend.py", line 4019, in batch_set_value
    x.assign(np.asarray(value, dtype=dtype_numpy(x)))
ValueError: Cannot assign value to variable ' BN-0-0/depthwise_kernel:0': Shape mismatch.The variable shape (1, 1, 64, 1), and the assigned value shape (3, 3, 64, 1) are incompatible.
[image]

AfifaIshtiaq, 2022-05-12:
> So you have a model with one one input or are doing the concatenation inside your model?
> 
> What do you mean by 'change dimension from 64 to 128'? Where did you make this change?

My first layer is concatenation layer which is included in supported layers then why I am having the following error?
opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/utils/generic_utils.py:497: CustomMaskWarning: Custom mask layers require a config and must override get_config. When loading, the custom mask layer must be passed to the custom_objects argument.
  category=CustomMaskWarning)
[VAI INFO] Update custom_layer_type: []

xieyi4650, 2022-07-21:
Hi @AfifaIshtiaq ,

could you upload the .h5 file, so we can debug it.

according to the log message, are there custom layers in the model?

renjiangyong, 2022-09-05:
Hi @AfifaIshtiaq， this issue is existed for long time, I will close this issue, If you have problem, you can reopen it.

closed 2022-09-05 by renjiangyong