Xilinx/Vitis-AI #797: is the VCK190 image correct?

giovanniguasti, 2022-05-11:
This problem was found [here](https://github.com/Xilinx/Vitis-AI/tree/master/Whole-App-Acceleration/apps/adas_detection)
the WAA ADAS example for VCK190 has this VCK190 image:
https://www.xilinx.com/member/forms/download/design-license-xef.html?filename=xilinx-vck190-dpu-v2020.2-v1.4.0.img.gz
but the new one for production silicon should be
https://www.xilinx.com/member/forms/download/design-license-xef.html?filename=xilinx-vck190-dpu-v2021.2-v2.0.0.img.gz

is this a typo?

sumitn-xilinx, 2022-05-13:
Not a typo, it's actually using the old version.
This is going to be fixed in VAI 2.5.

qianglin-xlnx, 2022-09-09:
Hi @giovanniguasti 
The issue is fixed in VAI2.5. Please refer to https://github.com/Xilinx/Vitis-AI/tree/master/examples/Whole-App-Acceleration/apps/adas_detection#1-setting-sd-card

closed 2022-09-09 by qianglin-xlnx