Xilinx/Vitis-AI #799: NaN in output - running Tensorflow2 classification with softmax on DPUCZDX8G

Orchidaceae, 2022-05-12:
## Summary
I have trained and successfully quantized and compiled a 3-layer MNIST classification model for DPUCZDX8G. But when I'm trying to run my model on the ZCU102 board I get unexpected NaNs and zeros in the classification output vector where only a few instances of the test images results in random float numbers but with very poor accuracy.

> Output examples:  
>
>...most of the output:
>>
>>[[nan  0. nan  0.  0. nan  0.  0.  0.  0.]]  
>
>...and in some instances:  
>>
>>[[2.9359882e-30 3.3529205e-04 2.2591801e-06 9.9949145e-01 2.2591801e-06
  1.5222234e-08 4.5376844e-05 9.3528639e-14 1.2334704e-04 8.5287086e-17]

**I had no issues in the previous deployment steps so what could cause this? Am I somehow interpreting the output wrong?**

## MNIST model definition
```python
 inputs = tf.keras.layers.Input(shape=(784,))
    dense = tf.keras.layers.Dense(128, activation='relu')(inputs)
    net = tf.keras.layers.Dense(10)(dense)
    prediction = tf.keras.layers.Activation('softmax')(net)
    float_model = tf.keras.Model(inputs=inputs, outputs=prediction, name='mnist_softmax_model')
```
```
Model: "mnist_softmax_model"
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
input_1 (InputLayer)         [(None, 784)]             0         
_________________________________________________________________
quant_input_1 (VitisQuantize (None, 784)               4         
_________________________________________________________________
quant_dense (QuantizeWrapper (None, 128)               100487    
_________________________________________________________________
quant_dense_relu (QuantizeWr (None, 128)               4         
_________________________________________________________________
quant_dense_1 (QuantizeWrapp (None, 10)                1300      
_________________________________________________________________
activation (Activation)      (None, 10)                0         
=================================================================
Total params: 101,795
Trainable params: 101,770
Non-trainable params: 25
_________________________________________________________________
```

## Evaluation of quantized model
```bash
2022-05-12 13:12:36.399445: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:185] None of the MLIR Optimization Passes are enabled (registered 2)
313/313 [==============================] - 1s 968us/step - loss: 0.2457 - sparse_top_k_categorical_accuracy: 0.9963
Evaluate on test data
79/79 [==============================] - 0s 2ms/step - loss: 0.2457 - sparse_top_k_categorical_accuracy: 0.9963
test loss, test acc: [0.24574635922908783, 0.9962999820709229]
Generate predictions for 3 samples
Prediction: 9   Truth:  9
Prediction: 2   Truth:  2
Prediction: 7   Truth:  7
```
## Compiling model for DPUCZDX8G
```bash
$ ARCH=/opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json
$ MODEL=quantized_models/mnist_MLP_softmax_q.h5
$ NAME=mnist_MLP_softmax_c
$ vai_c_tensorflow2 -m $MODEL -a $ARCH -o compiled_models/ -n $NAME
[INFO] Namespace(batchsize=1, inputs_shape=['1,784'], layout='NHWC', model_files=['quantized_models/mnist_MLP_softmax_q.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/mnist_MLP_softmax_c_org.xmodel', proto=None)
in_shapes: [[1, 784]]
[INFO] tensorflow2 model: /workspace/quantized_models/mnist_MLP_softmax_q.h5
[INFO] keras version: 2.6.0
[INFO] Tensorflow Keras model type: functional                      
[... 122 lines cut ...]
 10 
PREDICTION:	 0 	TRUTH:	 7
started graph runner
started graph runner
started graph runner
started graph runner
started graph runner
started graph runner
started graph runner
started graph runner
started graph runner
```
Only a few of all test images generate an output that does not contain zeros and nan, like for example image nr 34:
```bash
IMAGE NR  34
OUTPUT:
 [[3.6964852e-41 3.3333334e-01 3.3333334e-01 1.7476286e-22 1.0737801e-27
  3.3333334e-01 9.5417287e-21 2.6616346e-30 1.3799792e-08 4.3319380e-25]] 
RAW OUTPUT:
 TensorBuffer{@0xaaaafa467990,tensor=xir::Tensor{name = activation, type = FLOAT32, shape = {1, 10}},location=HOST_VIRT,data=[(Virt=0xaaaafc474630, 40)]} 
PRE OUTPUT SIZE:
 10 
PREDICTION:	 1 	TRUTH:	 7
```
The resulting accuracy of these results is very poor.

zz002, 2022-06-24:
Hi @Orchidaceae 
The input and output of the model are in fixed-point format, and the fixed-point position needs to be considered when writing and reading data.[src](https://github.com/Xilinx/Vitis-AI/blob/master/src/Vitis-AI-Runtime/VART/vart/dpu-runner/samples/resnet50_py_ext/resnet50_ext.py#L105)

qianglin-xlnx, 2022-07-27:
Hi @Orchidaceae 
Is this issue solved?

qianglin-xlnx, 2022-08-09:
Hi @Orchidaceae 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-08-09 by qianglin-xlnx