Xilinx/Vitis-AI #803: Segmentation fault during quantization with GPU

DeepTrial, 2022-05-18:
I have a resenet50 implemented by pytorch. When I quantize the model with device cuda, it fails with segmentation fault. However when switching to device cpu, it works. What should i do if i have to quantize with gpu?

[image]

DeepTrial, 2022-05-18:
Additional Information: Vitis-AI 2.0 with visit-ai-pytorch

zl200881, 2022-05-30:
Would you please disclose more running information, such as the warnings, errors, as well as command shells?

zl200881, 2022-06-01:
More information about your working environment?
Such as Ubuntu or windows, gpu or cpu, docker or conda env, graphics card?

niuxjxlnx, 2022-08-03:
@DeepTrial :

Please confirm GPU can be accessed and its driver works well. 
You can type "nvidia-smi" to check the log.  
A pytorch function "torch.cuda.device_count()" also can check GPU works or not.

niuxjxlnx, 2022-08-31:
Close since no response for a long time.

closed 2022-08-31 by niuxjxlnx