Xilinx/Vitis-AI #804: Xdputil fix.cpp Check failed  round_mode==DPU_mode

AfifaIshtiaq, 2022-05-18:
I'm able to compile my model and now I try to run xdputil on my model but it ran into an error. Although I can run it for other examples**

(vitis-ai-tensorflow2) Vitis-AI /workspace/models/ChEstModel > vai_c_tensorflow2 -m quantized_model.h5 -a /opt/vitis_ai/compiler/arch/DPUCVDX8G/VCK190/VCK190arch.json -o compile -n deeprx_compile
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['quantized_model.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/deeprx_compile_org.xmodel', proto=None)
[INFO] tensorflow2 model: /workspace/models/ChEstModel/quantized_model.h5
[INFO] keras version: 2.6.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 103/103 [00:00<00:00, 18796.26it/s]             
[INFO] infer shape (NHWC)  :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 176/176 [00:00<00:00, 2965.26it/s]              
[INFO] perform level-0 opt :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 66.53it/s]                    
[INFO] perform level-1 opt :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00, 280.69it/s]                   
[INFO] generate xmodel     :100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 176/176 [00:00<00:00, 3567.46it/s]              
[INFO] dump xmodel: /tmp/deeprx_compile_org.xmodel
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCVDX8G_ISA2_C64B5_0602001036088252
[UNILOG][INFO] Graph name: DeepRx, with op num: 420
[UNILOG][INFO] Begin to compile...
[UNILOG][INFO] Total device subgraph number 33, DPU subgraph number 15
[UNILOG][INFO] Compile done.
[UNILOG][INFO] The meta json is saved to "/workspace/models/ChEstModel/compile/meta.json"
[UNILOG][INFO] The compiled xmodel is saved to "/workspace/models/ChEstModel/compile/deeprx_compile.xmodel"
[UNILOG][INFO] The compiled xmodel's md5sum is c7bf4ec4c241d11c58bc3f47e1122829, and has been saved to "/workspace/models/ChEstModel/compile/md5sum.txt"

**Target**

root@xilinx-vck190-2021_2:~/models# xdputil benchmark resnet50.xmodel -i -1 1
XAIEFAL: INFO: Resource group Avail is created.
XAIEFAL: INFO: Resource group Static is created.
XAIEFAL: INFO: Resource group Generic is created.
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0517 08:39:34.758364  1172 test_dpu_runner_mt.cpp:475] shuffle results for batch...
I0517 08:39:34.759827  1172 performance_test.hpp:73] 0% ...
I0517 08:39:40.759963  1172 performance_test.hpp:76] 10% ...
I0517 08:39:46.760113  1172 performance_test.hpp:76] 20% ...
I0517 08:39:52.760265  1172 performance_test.hpp:76] 30% ...
I0517 08:39:58.760419  1172 performance_test.hpp:76] 40% ...
I0517 08:40:04.760569  1172 performance_test.hpp:76] 50% ...
I0517 08:40:10.760716  1172 performance_test.hpp:76] 60% ...
I0517 08:40:16.760866  1172 performance_test.hpp:76] 70% ...
I0517 08:40:22.761023  1172 performance_test.hpp:76] 80% ...
I0517 08:40:28.761174  1172 performance_test.hpp:76] 90% ...
I0517 08:40:34.761327  1172 performance_test.hpp:76] 100% ...
I0517 08:40:34.761395  1172 performance_test.hpp:79] stop and waiting for all threads terminated....
I0517 08:40:34.763039  1172 performance_test.hpp:85] thread-0 processes 142620 frames
I0517 08:40:34.763065  1172 performance_test.hpp:93] it takes 1652 us for shutdown
I0517 08:40:34.763082  1172 performance_test.hpp:94] FPS= 2376.87 number_of_frames= 142620 time= 60.0033 seconds.
I0517 08:40:34.763118  1172 performance_test.hpp:96] BYEBYE
Test PASS.
**root@xilinx-vck190-2021_2:~/models# xdputil benchmark deeprx_compile.xmodel -i -1 1
XAIEFAL: INFO: Resource group Avail is created.
XAIEFAL: INFO: Resource group Static is created.
XAIEFAL: INFO: Resource group Generic is created.
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0517 08:40:51.740269  1177 fix.cpp:32] Check failed: round_mode == "DPU_ROUND" (PY3_ROUND vs. DPU_ROUND) TODO:
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20:  1176 Aborted                 /usr/bin/python3 -m xdputil $***

oot@xilinx-vck190-2021_2:~/models#  xdputil benchmark deeprx_compile.xmodel 1
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0518 01:43:47.955123  1277 dpu_runner.cpp:206] [UNILOG][FATAL][VART_RUNNER_CONSTRUCTION_FAIL][Cannot create runner] Cannot find runner attr, this subgraph may not be compiled! subgraph name: subgraph_Raw-Channel-Est-In
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20:  1276 Aborted                 /usr/bin/python3 -m xdputil $*

zz002, 2022-05-19:
Hi @AfifaIshtiaq 
[fix.cpp](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Library/cpu_task/ops/fix/fix.cpp#L32) For the fix op, we only implement the DPU_ROUND mode, which does not support PY3_ROUND. You can add the implementation of PY3_ROUND to the source code, or use DPU_ROUND in the op.
There are logs when you compile the model: `Total device subgraph number 33, DPU subgraph number 15.`
There are a large number of unsupported subgraphs that interrupt the normal dpu subgraph.
`xdputil benchmark deeprx_compile.xmodel 1` .The default parameter `-i 1` is actually used when the `-i` parameter is not specified.
benchmark does not support cpu subgraph. subgraph_Raw-Channel-Est-In may be cpu subgraph.
You can use `xdputil xmodel -s test.svg your_xmodel`  to export your model topology, the red box is the cpu subgraph, you need to replace it with other supported ops or implement unsupported ops in the source code [e.g.](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Library/cpu_task/ops/nn_channel_shuffle/nn_channel_shuffle.cpp).

AfifaIshtiaq, 2022-05-19:
> Hi @AfifaIshtiaq [fix.cpp](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Library/cpu_task/ops/fix/fix.cpp#L32) For the fix op, we only implement the DPU_ROUND mode, which does not [...]
This is my svg graph Its not converting depthwise convolution to dpu for some blocks but convert for others. Can you tell me what can be the problem

[image]

AfifaIshtiaq, 2022-05-19:
> PY3_ROUND

I couldn't see any PY3_ROUND in my model.

AfifaIshtiaq, 2022-05-19:
> or use DPU_ROUND in the op

How can I use  or use DPU_ROUND in the op. In which command I have to mention it

wanghong4compiler, 2022-06-24:
Hi @AfifaIshtiaq ,

Which vitis ai version did you use? And could you share the model file so that I can reproduce this issue, thank you.

AfifaIshtiaq, 2022-06-24:
Hi I was able to resolve this issue to me it seemed to be related to the
fact that if we both cpu and dpu graphs in the model and xdputil works only
with dpu graph model. Please correct me if I’m wrong

qianglin-xlnx, 2022-07-05:
Hi @AfifaIshtiaq 
xdputil can work with the model which has both cpu and dpu graph.

closed 2022-07-05 by qianglin-xlnx