Xilinx/Vitis-AI #805: Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)

masommao, 2022-05-19:
use the command：python resnet18_quant.py --quant_mode calib --subset_len 256

zl200881, 2022-05-30:
Show more running logs?
Check the dataset and pre-trained model?

masommao, 2022-05-30:
as follow logs run on the GPU docker:
(my-ai-pytorch) Vitis-AI /workspace/tools/Vitis-AI-Quantizer/vai_q_pytorch/example > python resnet18_quant.py --quant_mode calib --subset_len 200

[VAIQ_NOTE]: Loading NNDCT kernels...
-------- Start resnet18 test 

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cuda'.

[VAIQ_NOTE]: =>Parsing ResNet...

[VAIQ_NOTE]: Start to trace model...

[VAIQ_NOTE]: Finish tracing.

[VAIQ_NOTE]: Processing ops...
██████████████████████████████████████████████████| 70/70 [00:00<00:00, 4281.52it/s, OpInfo: name 

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(quantize_result/ResNet.py)

[VAIQ_NOTE]: =>Get module with quantization.
  0%|                                                                       | 0/7 [00:00<?, ?it/s]
Segmentation fault (core dumped)

as follow logs run on the pc:
[VAIQ_NOTE]: Loading NNDCT kernels...
1373:INFO:57562-MainProcess: ================================================================================
1374:INFO:57562-MainProcess: Start of experiment: exp_real
1374:INFO:57562-MainProcess: vr3d
1374:INFO:57562-MainProcess: 2022-05-27 16:04:09
1374:INFO:57562-MainProcess: ================================================================================
1374:INFO:57562-MainProcess: Set seed to 42
*********Start pretrained_model_0523 quantization with optimization ***********
4615:INFO:57562-MainProcess: 
[VAIQ_NOTE]: Quantization calibration process start up...
4615:INFO:57562-MainProcess: 
[VAIQ_NOTE]: =>Quant Module is in 'cuda'.
4618:INFO:57562-MainProcess: 
[VAIQ_NOTE]: =>Parsing DispEdgeDecoders...
4927:INFO:57562-MainProcess: 
[VAIQ_NOTE]: Start to trace model...
/home/mason/workspace/Vitis-Ai-2.0/tools/Vitis-AI-Quantizer/vai_q_pytorch/my_demo/model/networks.py:202: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  assert(input.size(2) >= ref.size(2) and input.size(3) >= ref.size(3))
/home/mason/workspace/Vitis-Ai-2.0/tools/Vitis-AI-Quantizer/vai_q_pytorch/my_demo/model/networks.py:203: TracerWarning: Converting a tensor to a Python index might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  return input[:, :, :ref.size(2), :ref.size(3)]
5020:INFO:57562-MainProcess: 
[VAIQ_NOTE]: Finish tracing.
5168:INFO:57562-MainProcess: 
[VAIQ_NOTE]: Processing ops...
██████████████████████████████████████████████████| 137/137 [00:00<00:00, 6666.66it/s, OpInfo: name = 5206, type = add]
5301:INFO:57562-MainProcess: 
[VAIQ_NOTE]: =>Doing weights equalization...
5860:INFO:57562-MainProcess: 
[VAIQ_NOTE]: =>Quantizable module is generated.(quantize_result/DispEdgeDecoders.py)
6187:INFO:57562-MainProcess: 
[VAIQ_NOTE]: =>Get module with quantization.
/home/mason/software/anaconda3/install/envs/my_ai_quantizer/lib/python3.7/site-packages/torch/nn/_reduction.py:44: UserWarning: size_average and reduce args will be deprecated, please use reduction='mean' instead.
  warnings.warn(warning.format(ret))
  0%|          | 0/1 [00:00<?, ?it/s]128

Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)

masommao, 2022-05-30:
The configuration of my computer is as follows:

masommao, 2022-05-30:
The configuration of my computer is as follows:
2*RTX8000,memory:192G

zl200881, 2022-05-31:
It seems the process went  wrong when getting the module for quantization.
Can you identify the segmentation fault?  Like producing the core.dump file and viewing it using gdb?

masommao, 2022-05-31:
the error is running on GPU，When I use CPU is correct。

masommao, 2022-05-31:
the resnet18_quant.py is example on github，no change！and my demo is the same error。

zl200881, 2022-05-31:
Are you using docker or conda env? 
Make sure your working environment has installed the proper versions of python, pytorch, torchvision and other packages.

masommao, 2022-05-31:
yes，my env as follows：
(vitis-ai-pytorch) Vitis-AI /workspace > pip list
Package                       Version
----------------------------- ------------------------------
argon2-cffi                   21.1.0
async-generator               1.10
attrs                         21.4.0
backcall                      0.2.0
backports.functools-lru-cache 1.6.4
bleach                        5.0.0
certifi                       2021.5.30
cffi                          1.14.6
ck                            2.6.1
colorama                      0.4.4
cycler                        0.11.0
dataclasses                   0.8
decorator                     5.1.1
defusedxml                    0.7.1
dill                          0.3.4
entrypoints                   0.4
graphviz                      0.16
h5py                          2.10.0
importlib-metadata            4.8.1
importlib-resources           5.4.0
iniconfig                     1.1.1
ipykernel                     5.5.5
ipython                       7.16.1
ipython-genutils              0.2.0
ipywidgets                    7.6.5
jedi                          0.17.2
Jinja2                        3.0.3
joblib                        1.1.0
jsonschema                    4.1.2
jupyter                       1.0.0
jupyter-client                7.1.2
jupyter-console               6.4.3
jupyter-core                  4.8.1
jupyterlab-pygments           0.1.2
jupyterlab-widgets            1.1.0
kiwisolver                    1.3.1
MarkupSafe                    2.0.1
marshmallow                   3.15.0
matplotlib                    3.3.4
mistune                       0.8.4
more-itertools                8.12.0
nbclient                      0.5.9
nbconvert                     6.0.7
nbformat                      5.1.3
nest-asyncio                  1.5.5
ninja                         1.10.2.3
notebook                      6.3.0
numpy                         1.17.2
olefile                       0.46
opencv-contrib-python         4.5.5.64
orderedset                    2.0.3
packaging                     21.3
pandas                        1.1.5
pandocfilters                 1.5.0
parso                         0.7.1
pexpect                       4.8.0
pickleshare                   0.7.5
Pillow                        8.1.0
pip                           21.3.1
pluggy                        1.0.0
progressbar2                  3.55.0
prometheus-client             0.14.1
prompt-toolkit                3.0.29
protobuf                      3.11.4
ptyprocess                    0.7.0
py                            1.11.0
pybind11                      2.8.0
pybind11-global               2.8.0
pycparser                     2.21
pydot                         1.4.2
Pygments                      2.12.0
pyparsing                     3.0.8
pyrsistent                    0.17.3
pytest                        6.2.5
python-dateutil               2.8.2
python-utils                  3.1.0
pytorch-nndct                 0.1.0-996459f-torch1.7.1-cu101
pytz                          2022.1
PyYAML                        5.4.1
pyzmq                         22.3.0
qtconsole                     5.2.2
QtPy                          2.0.1
SC                            1.4
scikit-learn                  0.24.1
scipy                         1.3.1
Send2Trash                    1.8.0
setuptools                    59.6.0
six                           1.16.0
sklearn                       0.0
terminado                     0.12.1
testpath                      0.6.0
threadpoolctl                 3.1.0
toml                          0.10.2
torch                         1.7.1+cu101
torchvision                   0.8.2+cu101
tornado                       6.1
tqdm                          4.64.0
traitlets                     4.3.3
typing_extensions             4.1.1
wcwidth                       0.2.5
webencodings                  0.5.1
wheel                         0.37.1
widgetsnbextension            3.5.1
xnnc                          1.4.0
zipp                          3.6.0

masommao, 2022-05-31:
I only config the pathes of the “data_dir” and “model_dir”，code as follows：

import os
import re
import sys
import argparse
import time
import pdb
import random
from pytorch_nndct.apis import torch_quantizer
import torch
import torchvision
import torchvision.transforms as transforms
from torchvision.models.resnet import resnet18

from tqdm import tqdm
os.environ["CUDA_HOME"] = "/usr/local/cuda"
#device = torch.device("cuda")
#device = torch.device("cpu")
device = torch.device("cuda" if torch.cuda.is_available() else "cpu")

parser = argparse.ArgumentParser()
#"/workspace/dataSets/imageNet"
#"/home/mason/workspace/Vitis-AI_2.0/dataSets/imageNet"
parser.add_argument(
    '--data_dir',
    default="/workspace/dataSets/imageNet",
    help='Data set directory, when quant_mode=calib, it is for calibration, while quant_mode=test it is for evaluation')
parser.add_argument(
    '--model_dir',
    default="./pre_model/",
    help='Trained model file path. Download pretrained model from the following url and put it in model_dir specified path: https://download.pytorch.org/models/resnet18-5c106cde.pth'
)
parser.add_argument(
    '--subset_len',
    default=200,
    type=int,
    help='subset_len to evaluate model, using the whole validation dataset if it is not set')
parser.add_argument(
    '--batch_size',
    default=32,
    type=int,
    help='input data batch size to evaluate model')
parser.add_argument('--quant_mode', 
    default='calib',
    choices=['float', 'calib', 'test'], 
    help='quantization mode. 0: no quantization, evaluate float model, calib: quantize, test: evaluate quantized model')
parser.add_argument('--fast_finetune', 
    dest='fast_finetune',
    action='store_true',
    help='fast finetune model before calibration')
parser.add_argument('--deploy', 
    dest='deploy',
    action='store_true',
    help='export xmodel for deployment')
args, _ = parser.parse_known_args()

def load_data(train=True,
              data_dir='dataset/imagenet',
              batch_size=128,
              subset_len=None,
              sample_method='random',
              distributed=False,#distributed分布式
              model_name='resnet18',
              **kwargs):

  #prepare data
  # random.seed(12345)
  traindir = data_dir + '/train'
  valdir = data_dir + '/val'
[... 194 lines cut ...]

if __name__ == '__main__':

  model_name = 'resnet18'
  file_path = os.path.join(args.model_dir, model_name + '.pth')

  feature_test = ' float model evaluation'
  if args.quant_mode != 'float':
    feature_test = ' quantization'
    # force to merge BN with CONV for better quantization accuracy
    args.optimize = 1
    feature_test += ' with optimization'
  else:
    feature_test = ' float model evaluation'
  title = model_name + feature_test

  print("-------- Start {} test ".format(model_name))

  # calibration or evaluation
  quantization(
      title=title,
      model_name=model_name,
      file_path=file_path)

  print("-------- End of {} test ".format(model_name))

zl200881, 2022-05-31:
I reruned your code in my computer and it completed successfully, no mater in CPU or GPU.
My conda env is python3.7.12, torch 1.11.0+cu102, torchvision 0.12.0+cu102
Try upgrading your torch and torchvision?

masommao, 2022-06-01:
code runed on docker 、 Ubuntu or win10？

zl200881, 2022-06-01:
Not docker, a self-built conda env
Ubuntu

masommao, 2022-06-02:
I found that  Running with command is no error，use pycharm IDE have this error。but Running with command in docker have the same error。

masommao, 2022-06-02:
I found that Running with command is no error on Ubuntu，use pycharm IDE have this error。but Running with command in docker have the same error。

Docker Image Version:  1.4.1.978 
Vitis AI Git Hash:  c6d79ee83 
Build Date: 2022-04-27

Vitis-AI /workspace/tools/Vitis-AI-Quantizer/vai_q_pytorch/example > python resnet18_quant.py --quant_mode calib --subset_len 200
**************************

[VAIQ_NOTE]: Loading NNDCT kernels...
###########################
-------- Start resnet18 test 

[VAIQ_NOTE]: Quantization calibration process start up...

[VAIQ_NOTE]: =>Quant Module is in 'cuda'.

[VAIQ_NOTE]: =>Parsing ResNet...

[VAIQ_NOTE]: Start to trace model...

[VAIQ_NOTE]: Finish tracing.

[VAIQ_NOTE]: Processing ops...
██████████████████████████████████████████████████| 70/70 [00:00<00:00, 2105.50it/s, OpInfo

[VAIQ_NOTE]: =>Doing weights equalization...

[VAIQ_NOTE]: =>Quantizable module is generated.(quantize_result/ResNet.py)

[VAIQ_NOTE]: =>Get module with quantization.
  0%|                                                                | 0/7 [00:00<?, ?it/s]Segmentation fault (core dumped)

Segmentation fault (core dumped)

zl200881, 2022-06-14:
Try debugging your code using pdb to find the crash line.

niuxjxlnx, 2022-08-03:
@masommao :

Please confirm GPU can be accessed and its driver works well. 
You can type "nvidia-smi" to check the log.  
A pytorch function "torch.cuda.device_count()" also can check GPU works or not.

niuxjxlnx, 2022-08-31:
Close it since no response for a long time.

closed 2022-08-31 by niuxjxlnx