Xilinx/Vitis-AI #806: "No LSB modules are available" when running "./build.sh" in Vitis AI Docker

dylangeon, 2022-05-19:
Hello, Xilinx experts:
I am using the the Vitis-AI example from this link: https://github.com/Xilinx/Vitis-AI/tree/master/examples/DPUCADF8H/tf_inception_v1
and when I run "./build.sh" in Vitis AI Docker, I got an error message "No LSB modules are available".
If I run "lsb_release -a" in the docker, I got:
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic

However, outside the docker, when I run "lsb_release -a", I got:
LSB Version:	:core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID:	RedHatEnterpriseServer
Description:	Red Hat Enterprise Linux Server release 7.9 (Maipo)
Release:	7.9
Codename:	Maipo

I also tried to install lsb-core, by running "sudo apt-get install lsb-core", I got:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package lsb-core

I am very confused, I got different linux system outside and inside the docker, and I can 't install lsb-core in the docker. Anybody could help? Thank you very much!!

wanghy-xlnx, 2022-06-23:
Hi @dylangeon ,

Docker is similar to a virtual machine, although RedHat Enterprise 7.9 is running on your host, but Ubuntu 18.04 is running inside docker. But docker is not equivalent to a real virtual machine and does not support all commands.

You can ignore the error message "No LSB modules are available". The "./build.sh" script is just to get the correct OS and version number and it has worked. The test is as follows:

[image]

dylangeon, 2022-06-23:
Thank you very much Hanxue and wanghy-xlnx for your help!
wanghy-xlnx,
You answered my question! Really appreciate it.
Dylan

closed 2022-07-13 by qianglin-xlnx