Xilinx/Vitis-AI #808: How to preprocess image for yolovx on ZCU102

Gang-Chen-China, 2022-05-20:
Hi, When I run yolovx.xmodel ZCU102，the command is:

` sample : ./test_jpeg_yolovx  yolo.xmodel test.jpg`

it outputs negative values，should I need to preprocess image as folloing: 

```  
data channel order: RGB(0~255)                  
input size: h * w = 640 * 640
mean = (0.0, 0.0, 0.0)
std = (1.0, 1.0, 1.0)
input = (input * normalize  - mean) / std
```

zz002, 2022-05-25:
hi @Gang-Chen-China 
The preprocessing of our model is
`data channel order: BGR(0~255)   
input = letterbox(input)
input = (input - mean) * scale `
[preprocessing](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Library/yolovx/src/yolovx_imp.cpp#L26)
If you want to use test_jpeg_yolovx, you also need to add a configuration file as follows, and put it in the same directory as your model, similar to the following directory structure
[post-processing](https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Library/xnnpp/src/yolovx.cpp#L69)
`cat tsd_yolox_pt.prototxt
model {
    kernel {
        mean:  0
        mean:  0
        mean:  0
        scale: 1
        scale: 1
        scale: 1
    }
    yolo_v5_param {
        stride:32
        stride:16
        stride:8
        yolo_param {
            num_classes: 45
            anchorCnt: 1
            layer_name: "_2__"
            layer_name: "_1__"
            layer_name: "_0__"
            conf_threshold: 0.25
            nms_threshold: 0.45
        }
    }
`
`tsd_yolox_pt/
|-- tsd_yolox_pt.prototxt
|-- tsd_yolox_pt.xmodel
`

qianglin-xlnx, 2022-07-05:
hi @Gang-Chen-China 
Is this issue solved?

closed 2022-07-12 by qianglin-xlnx

jzymessi, 2022-08-02:
@qianglin-xlnx :
     I also deploy yolox model, I want to ask  prototxt files in the scheme, scale, stride, anchorCnt, layer_name value is how to determine?

qianglin-xlnx, 2022-08-02:
HI @jzymessi 
They are determined by the float model. You can ask the float model provider who design and train the model.