Xilinx/Vitis-AI #810: port_profile in dpu.cpp

shikhagoel003, 2022-05-23:
Hi,

I am writing this request for Vitis version 1.3.1. 
I want to know port profiles like how many read writes are there on each port - HP0, HP1, HP3 and HP3.
I have gone through the DPU driver code given in tools/Vitis-AI-Runtime/DNNDK/n2cube/src. I am looking at file dpu.cpp. I find the function "print_task_port_profile" in line 2542. I see that it can print the read and writes on the port. When I recompile the driver code and rerun it on the FPGA board, this function returns me all read and writes as 0. 
Can you look at this issue? I want to know the profiling details of each of the ports.

Thanks
Shikha

wanghy-xlnx, 2022-06-17:
Hi @shikhagoel003 ,

Unfortunately, this function is now invalid, because after calling the DPU, The following struct members are not assigned in the driver.
```
struct port_profile_t {
    unsigned int       port_hp0_read_byte;     /* bytes read in HP0 port (128-bit) */
    unsigned int       port_hp0_write_byte;    /* bytes write in HP0 port (128-bit) */

    unsigned int       port_hp1_read_byte;     /* bytes read in HP1 port (128-bit) */
    unsigned int       port_hp1_write_byte;    /* bytes write in HP1 port (128-bit) */

    unsigned int       port_hp2_read_byte;     /* bytes read in HP2 port (128-bit) */
    unsigned int       port_hp2_write_byte;    /* bytes write in HP2 port (128-bit) */

    unsigned int       port_hp3_read_byte;     /* bytes read in HP3 port (128-bit) */
    unsigned int       port_hp3_write_byte;    /* bytes write in HP3 port (128-bit) */

    unsigned int       port_gp_read_byte;      /* bytes read in GP port (32-bit for DPU inst) */
    unsigned int       port_gp_write_byte;     /* bytes write in GP port (32-bit for write profiler) */

    unsigned long long dpu_cycle;              /* the start timestamp before running (RETURNED) */
};
```

The dpu_launch_execution_session call is used to schedule the DPU.  For example, in [dpu.cpp](https://github.com/Xilinx/Vitis-AI/blob/1.3.1/tools/Vitis-AI-Runtime/DNNDK/n2cube/src/dpu.cpp#L1369), when the DPU scheduling is completed, the following variables are assigned, but there are no variables mentioned above.
```
task->node_list[i]->time_start = aol_run.time_start;
task->node_list[i]->time_end = aol_run.time_end;
task->node_list[i]->coreID = mask2id(aol_run.core_mask);
```

qianglin-xlnx, 2022-07-27:
Hi @shikhagoel003 
Is this issue solved?

qianglin-xlnx, 2022-08-09:
Hi @shikhagoel003 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-08-09 by qianglin-xlnx