Xilinx/Vitis-AI #811: Quantization error YOLOv5 using pytorch

JaviMota, 2022-05-23:
Hi 

Im trying to quantize and compile the yolov5s but Im getting an error when I do the following steps:

model = torch.hub.load('ultralytics/yolov5', 'yolov5s', pretrained=True, autoshape=False)

rand_in = torch.randn([100, 3, 640, 640])
quantizer = torch_quantizer("test", model, (rand_in), output_dir=".")
quantized_model = quantizer.quant_model
#quantizer.export_quant_config()
quantizer.export_xmodel(deploy_check=False, output_dir=".")

Before the quant_mode "test" I have done the quant_mode "calib" and it works fine.
With the quant_mode "test" Im geting an error in the export_xmodel step saying:
"TypeError: 'data' must be a numpy ndarray"

[image]

It seems is trying to convert a None value to a ndarray in the qproc/utils.py file.

Im following more or less the steps ultralytics is doing for converting to onnx, tflite... to try to get the xmodel.

In https://github.com/Xilinx/Vitis-AI/issues/675 you mention it is possible to compile yolov5 for the DPU.

Can you please give some tips to get the yolov5 xmodel using pytorch?

Thank you.

Best regards,
Javi

zl200881, 2022-05-31:
You may not have the forward computation as the function _evaluate()_ in the example resnet18_quant.py does.

zl200881, 2022-06-01:
Your code can be modified as folows:

model = torch.hub.load('ultralytics/yolov5', 'yolov5s', pretrained=True, autoshape=False)

rand_in = torch.randn([100, 3, 640, 640])
quantizer = torch_quantizer("test", model, (rand_in), output_dir=".")
quantized_model = quantizer.quant_model
**acc1_gen, acc5_gen, loss_gen = evaluate(quant_model, val_loader, loss_fn) # adding forward computation**
#quantizer.export_quant_config()
quantizer.export_xmodel(deploy_check=False, output_dir=".")

JaviMota, 2022-06-03:
Hi

I have done the changes and still having the same issue

import torch
from pytorch_nndct.apis import torch_quantizer

def evaluate(model):
    model.eval()
    model = model.to("cpu")

    return

model = torch.hub.load('ultralytics/yolov5', 'yolov5s', pretrained=True, autoshape=False)
model = model.cpu()
rand_in = torch.randn([100, 3, 640, 640])
quantizer = torch_quantizer("test", model, (rand_in), output_dir=".")
quantized_model = quantizer.quant_model
evaluate(quantized_model)
#quantizer.export_quant_config()
quantizer.export_xmodel(deploy_check=False)

Traceback (most recent call last):
  File "/opt/.pycharm_helpers/pydev/pydevd.py", line 1491, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/opt/.pycharm_helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/opt/project/main.py", line 18, in <module>
    quantizer.export_xmodel(deploy_check=False)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 110, in export_xmodel
    self.processor.export_xmodel(output_dir, deploy_check)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 180, in export_xmodel
    dump_xmodel(output_dir, deploy_check, self._lstm_app)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 203, in dump_xmodel
    deploy_graphs = get_deploy_graph_list(quantizer.quant_model, quantizer.Nndctgraph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 290, in get_deploy_graph_list
    update_nndct_blob_data(quant_model, g_optmizer.frozen_graph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 165, in update_nndct_blob_data
    dfs(source, visited)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 156, in dfs
    dfs(cn, visited)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 156, in dfs
    dfs(cn, visited)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 152, in dfs
    node.out_tensors[0].from_ndarray(data)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_tensor.py", line 66, in from_ndarray
    raise TypeError("'data' must be a numpy ndarray")
TypeError: 'data' must be a numpy ndarray

zl200881, 2022-06-15:
@JaviMota Can you attach your code file rather than pasting your code? Then I can run your code without change.

JaviMota, 2022-06-15:
I cannot attach the script here but I send you a link to download it.

https://drive.google.com/file/d/1I9quRyJqWBHd_QUbrXxqphr3QWZASXJI/view?usp=sharing

Thanks!

zl200881, 2022-06-15:
@JaviMota Can you post your log or error information? I don't understand your problem. 
When your quant_mode is "test" and your model is to be deployed, the batchsize shall be set 1. And if your batchsize>1, it will be reset as 1.

JaviMota, 2022-06-15:
The issue I have is:
- First I run the script using --quant_mode "calib" and --batchsize 32. It generates 3 files: bias_corr.pth, DetectMultiBackend.py and quant_info.json. It works fine.
- Secondly, I run the script again but this time with --quant_mode "test". With this mode inside the evaluate function in  "output = model(data)" it complains about the error I mentioned "KeyError: DetectMultiBackend::DetectMultiBackend/Model[model]/Detect[model]/Detect[24]/Conv2d[m]/ModuleList[0]/10593". If I force it to batchsize=2 it works but in "quantizer.export_xmodel(deploy_check=False, output_dir=quant_model)" it complains that batchsize=1 is needed.

Im using the default vitis-ai-pytorch, not sure if the yolov5 if fully compatible in this environment.

zl200881, 2022-06-16:
@JaviMota (1) Can you show more lines about the "KeyError"?
(2) If possible, please attach all ·necessary files so that I can reproduce your results. The files include quantize.py and ./files/instances_val2017.json.

JaviMota, 2022-06-16:
[image]

https://drive.google.com/drive/folders/1ecmmgD-8M7KHGuUOCWHNCm9moXIKfi5j?usp=sharing

zl200881, 2022-06-17:
@JaviMota Do you use the same quantize.py ( and imported model file) without any change for the quant_mode "calib" and "test" ? 
Nothing should be changed for test after calibration.

JaviMota, 2022-06-17:
Hi 
1. Yes I run the same script two times, first with "calib" and after with "test".
2. What error you get? Are you running it in Vitis-AI 2.0 docker using pytorch environment? I have uploaded again the quantize.py just in case. Im not getting any gramma exceptions.
https://drive.google.com/drive/folders/1ecmmgD-8M7KHGuUOCWHNCm9moXIKfi5j?usp=sharing

zl200881, 2022-06-20:
@JaviMota Two errors appear when calibrating the float model using your code and dataset:
1. "test_transform" in quantize.py not defined before using
2. data size error: 
[image]

JaviMota, 2022-06-20:
Ok sorry, seems you need to replace your common.py file, where is the test_transform function.
I have uploaded the file in this link:
https://drive.google.com/drive/folders/1ecmmgD-8M7KHGuUOCWHNCm9moXIKfi5j?usp=sharing

zl200881, 2022-06-22:
@JaviMota I ran your code in "calib" mode, and got the quantized results (bias_corr.pth, DetectMultiBackend.py, quant_info.json) as well as the warning information that some operators (e.g. "aten::pow", "strided_slice_inplace_copy" and "aten::silu_") were not supported when quantizing. 
Then I ran the code in "test" model and got a crush as follows. It seems the model can not be quantized because of some operators. 
[image]

JaviMota, 2022-06-22:
Ok @zl200881, thanks for your time. I was trying to compile it because https://github.com/Xilinx/Vitis-AI/issues/675#issuecomment-1085362536 said yolov5 is supported in Vitis-AI. But if some operators are not compatible I will not be able to compile it for the DPU.

zl200881, 2022-06-28:
@JaviMota Hi, for the unsupported operators in your model, you can modify them to fit Vitis-AI by referring to https://github.com/Xilinx/Vitis-AI/blob/master/model_zoo/model-list/pt_yolox_TT100K_640_640_73G_2.5/model.yaml.

closed 2022-08-31 by niuxjxlnx

githubcyy, 2022-09-01:
Hi, everyone. I'm a newer about vitis ai, I want to use yolov5 to quantization, which tutorial to follow? Thanks!

zl200881, 2022-09-01:
@githubcyy Hi, please follow the steps in https://github.com/Xilinx/Vitis-AI to set working environments, and follow [https://docs.xilinx.com/r/2.0-English/ug1414-vitis-ai/Overview](Vitis AI User Guide) to quantize your model.

githubcyy, 2022-09-02:
@zl200881 Thanks! When preparing files for vai_q_pytorch, I have model.pth and calibration dataset, but I don't have model.py. Do you have any relevant examples?
[image]
And I don't find "An example is available in [Vitis AI GitHub](https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Quantizer/vai_q_pytorch/example/resnet18_quant.py)", Has this link changed?

zl200881, 2022-09-02:
@githubcyy:
The link does not change.
It is right [here](https://github.com/Xilinx/Vitis-AI/tree/master/src/Vitis-AI-Quantizer/vai_q_pytorch/example).

Fomalhautwx, 2023-01-18:
@githubcyy The link hasn't changed, but in a former version of the repo. You need to switch to older versions to find it.