Xilinx/Vitis-AI #813: VCK190 DPU limitation: Depth-wise convolution

AfifaIshtiaq, 2022-05-23:
While compiling my model I found out that
**depthwise-conv2d** is supported by **DPU** if **dilation** 1,1 **Kernel** 3,3 pad 1,1,1,1 stride 1,1
But  **depthwise-conv2d** is mapped to **CPU** if  **dilation** 3,2 **Kernel** 3,3 pad 0,0,0,0 stride 1,1

I'm confused by the dcoumentation where here they say **depthwise-conv2d** is not supported by DPU https://docs.xilinx.com/r/1.3-English/ug1414-vitis-ai/Supported-OPs-and-DPU-Limitations but from the compiled results I can see some get maped

Also in here they say **depthwise-conv2d** can be  supported by DPU https://docs.xilinx.com/r/1.3-English/ug1414-vitis-ai/Operators-Supported-by-TensorFlow 

Here is the svg graph
[image]

wanghong4compiler, 2022-05-24:
Hi @AfifaIshtiaq ，
for now, we only support depthwise conv with dilation_h == dilation_w.

AfifaIshtiaq, 2022-05-30:
> Hi @AfifaIshtiaq ， for now, we only support depthwise conv with dilation_h == dilation_w.

Thanks, @wanghong4compiler I resolved this issue Can you also let me know the reason behind CPU mapping of the output convolution layer in the upper graph?

[image]

wanghong4compiler, 2022-07-13:
Hi @AfifaIshtiaq ,
DPU's data result are all XINT8, the last fix2fload op would transfer the fixed XINT8 result float back, this will keep the whole model equivalent.

closed 2022-07-13 by qianglin-xlnx