Xilinx/Vitis-AI #815: Is there a way to use VART in python after a Vivado DPU integration ?

just-an-intern, 2022-05-25:
Hello,

I would like to use VART in python after the DPU integration from the **Vivado flow**.
After the installation of VART following this [step ](https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Installing-Vitis-AI-Runtime-on-the-Evaluation-Board) from the UG1414 guide, VART requires the dpu.xclbin location, that is not created through the [Vivado flow](https://github.com/Xilinx/Vitis-AI/blob/master/dsa/DPU-TRD/prj/Vivado/README.md). So the output of the show_dpu command is "Check failure stack trace:" error and I cannot configure a location of the xclbin in the vart.conf file.

Is there a way to use the VART python library with the DPU integrated through the Vivado flow ?
Thank you for your help

zz002, 2022-05-30:
hi @just-an-intern 
You need to compile vitis ai from source in Vivado flow's sdk. The distribution package is not compiled for vivado.

https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Runtime/VART/quick_start_for_edge.md#setting-up-the-host
https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Library#setting-up-the-host

just-an-intern, 2022-05-31:
> hi @just-an-intern You need to compile vitis ai from source in Vivado flow's sdk. The distribution package is not compiled for vivado. [...]

Hi @zz002 ,
I tried recompiling the vitis-ai Runtime and loaded it in my FPGA (following the steps that you gave me), but it creates another issue. Now I have an error when using the python library (xir not having the attribute Graph) and it seems that a vart.conf file is created.
Which are the compilation option that I should set ?

Thank you for your help

zz002, 2022-06-02:
```
xir --build-python 
vart --build-python --cmake-options='-DENABLE_DPU_RUNNER=ON' --cmake-options='-DENABLE_SIM_RUNNER=ON' --cmake-options='-DENABLE_CPU_RUNNER=ON'   
Vitis-AI-Library --build-python 
```
If you don't need to debug vitis ai code, please add`--type=release`

just-an-intern, 2022-06-07:
Hi @zz002 ,

Thank you for your response and support. I was able to get the DPU to work using the Vitis flow(I had trouble with it initially, which is why I used the Vivado flow) and using the python library. I will try your solution later.

TitechTraj, 2022-06-23:
Hi, I am stuck with the same issue to run python VART  with Vivado flow.

So I tried the approach of @zz002 
> ```
> xir --build-python 
> vart --build-python --cmake-options='-DENABLE_DPU_RUNNER=ON' --cmake-options='-DENABLE_SIM_RUNNER=ON' --cmake-options='-DENABLE_CPU_RUNNER=ON'   
> Vitis-AI-Library --build-python 
> ```
> 
> If you don't need to debug vitis ai code, please add`--type=release`

This might seem a basic question, but how do I use these built libraries? 
Where or how should I load them in my FPGA?

Thanks for the help

zz002, 2022-06-23:
Hi @TitechTraj 
After compiling, you will see the installation directory similar to the following, copy Debug/* to /usr/* of your board
```
-- Up-to-date: /opt/petalinux/2022.1/sysroots/cortexa72-cortexa53-xilinx-linux/install/Debug/...
-- Up-to-date: /opt/petalinux/2022.1/sysroots/cortexa72-cortexa53-xilinx-linux/install/Debug/...
```

TitechTraj, 2022-06-23:
Thanks a lot for the quick response!

I was focused on the ~/build folder as mentioned by the VART compilation guide and I didn't understand what to do with all those files, but this looks more structured now. 

I will try and see if it works now.

TitechTraj, 2022-06-23:
Unfortunately, this did not solve the issue.  VART is still looking for the dpu.xclbin file.

```
F0614 15:10:03.262403  1795 xrt_bin_stream.cpp:59] Check failed: fd_ > 0 (-1 vs. 0) , open(/usr/lib/dpu.xclbin) failed.
*** Check failure stack trace: ***
```
I am a bit confused between the Vitis and Vivado flows. So for Vitis, it creates this xclbin file which the XRT uses to identify the DPU. For Vivado, we have to cross-compile models by linking the VART libraries when using C++?
But it looks like the python runtime follows the vitis flow and looks for the xclbin file?

zz002, 2022-06-23:
@TitechTraj 
Execute the following instructions, you can send the information directly to my mailbox，zhenze.wang@amd.com
`env DEBUG_DPU_CONTROLLER=1 xdputil query`

zz002, 2022-08-01:
@TitechTraj According to the communication in the email, this issue has been resolved, we will close this issue.

closed 2022-08-01 by qianglin-xlnx

Shreyas-NR, 2024-01-12:
@zz002 @TitechTraj 
Can you share your email conversation here? so it is helpful for other too!