Xilinx/Vitis-AI #816: GCP error when buld GPU docker

Zhijiexu-nuro, 2022-05-26:
Hi,
when I run ./docker_build_gpu.sh, I meet the GCP error.
W: GPG error: https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY A4B469963BF863CC
E: The repository 'https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  InRelease' is not signed.
and how to solve it
Thanks,

zhzhqian, 2022-06-01:
same issue, have no idea what to do

janifer112x, 2022-09-06:
Hi @zhzhqian，
this issue has been addressed in the latest commit of mater branch, please have a try and let us know if you still have any problem. I will close this issue for now. feel free to reopen it  if you have any concern.

closed 2022-09-06 by janifer112x