Xilinx/Vitis-AI #820: Custom Board - DPU Integration Vitis Flow

just-an-intern, 2022-05-31:
Hi,

I tried integrating the DPU B4096 on a custom board (from the supplier Enclustra, board PE1 and SOM XU5 with a Zynq Ultrascale+ - ZU4EV) but **I have an error** when running any CNN with the python api.

Software that I am using:
- Vivado / Vitis 2020.2
- Petalinux 2020.2
- Vitis-AI 1.3.2

I am using the Vitis [flow](https://github.com/Xilinx/Vitis-AI/tree/1.3.2/dsa/DPU-TRD/prj/Vitis) after a platform creation using [this tutorial](https://xilinx.github.io/Vitis-Tutorials/2020-2/docs/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/README.html).
I included the [meta-vitis-ai](https://github.com/Xilinx/meta-vitis-ai) for 1.3.2 and used the recipes for my board, provided by Enclustra.

Using lsmod, I can see the components : dmaproxy, mali, zocl and xbutil query shows me my DPU B4096.

However when I try the simple [MNIST tutorial](https://github.com/Xilinx/Vitis-AI-Tutorials/tree/1.3/Design_Tutorials/09-mnist_pyt-master) (using the right arch.json file). I have the following output:
```    
WARNING: Logging before InitGoogleLogging() is written to STDERR
W0512 10:56:18.172255 13735 xrt_cu.cpp:195] cu timeout! device_core_idx 0  handle=0xaaaae87954d0 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10009  bo=1 is_done 0
I0512 10:56:18.172366 13735 xrt_cu.cpp:105] Total: 10009960us   ToDriver: 24us  ToCU: 18446741814639326us       Complete: 0us   Done: 2269080162us
F0512 10:56:18.172405 13735 dpu_control_xrt_edge.cpp:177] dpu timeout! core_idx = 0
LSTART 154  LEND 154  CSTART 23  CEND 23  SSTART 1  SEND 1  MSTART 0  MEND 0  CYCLE_L 6172  CYCLE_H 0
*** Check failure stack trace: ***
Aborted
```
and the following error in the kernel
`[drm:zocl_client_release [zocl]] *ERROR* Please investigate stale cmds`

I read on [this implementation (LFAR73058)](https://support.xilinx.com/s/article/73058?language=en_US) on the Zedboard, that the clock for the DPU can depend on the device and available power. How can I make sure that the issue is not due to that factor ?

Thank you for your help

imzongjian, 2022-06-05:
I met the same error here,.My tools Version: vitis ai 2.0, petalinux/vitis 2021.2 
And I would meet this after I successfully inference once and only once.

tianfang-fafafa, 2022-06-28:
hi @just-an-intern ,  could you please check system interrupt first ? Dpu will time out if zocl-CU interrupt configure is incorrected

just-an-intern, 2022-06-29:
> hi @just-an-intern , could you please check system interrupt first ? Dpu will time out if zocl-CU interrupt configure is incorrected

Hi @tianfang-fafafa , 
Thank you for your answer.
I rebuilt my Vitis-AI sources / petalinux using the Makefile for the [plateform creation](https://github.com/Xilinx/Vitis-Tutorials/tree/2020.2/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104) and customizing the environment for my board and adding the petalinux recipes for my board and I successfully ran the MNIST Tutorial using two B2304 cores and the resnet50 classification example.
However I struggle to use these cores using my custom models (custom [RetinaFace](https://github.com/Xilinx/Vitis-AI/issues/886) and a face feature extractor).

But I am still interested in knowing how we can choose the cores for a specific UltraScale MPSoc. Is there power specification for the Board ? And for a given MPSoc (here ZU4EV), is the core choice only limited by the available BRAM and URAM  ? Or is there other factor ?
I still did not try to integrate another larger DPU (due to time constraints).

marenan, 2022-08-30:
> Hi @tianfang-fafafa , Thank you for your answer. I rebuilt my Vitis-AI sources / petalinux using the Makefile for the [plateform [...]

Hi, you succed quantizing your retinaface model with vai_q_pytorch？

just-an-intern, 2022-08-30:
> Hi, you succed quantizing your retinaface model with vai_q_pytorch？

Hi @marenan , I succeded in quantizing my RetinaFace model, after fixing multiple issues, such as replacing LeakyReLUs to ReLUs, and fixing some errors in my scripts.

I will close this issue, as I succeded in integrating my DPU and using my model.
Thank you all for your help

closed 2022-08-30 by just-an-intern

marenan, 2022-08-30:
> Hi @marenan , I succeded in quantizing my RetinaFace model, after fixing multiple issues, such as replacing LeakyReLUs to ReLUs, and fixing some errors in my scripts.
>
> I will close this issue, as I [...]

Could you give more advices or sharing your quantization scripts? I am always struggling with the tracing state of quantization, thanks!

just-an-intern, 2022-08-31:
Hi @marenan , 
For the quantization, I simply used the [quantization script](https://github.com/Xilinx/Vitis-AI-Tutorials/blob/1.4/Design_Tutorials/09-mnist_pyt/files/quantize.py) from the MNIST tutorial (and simply replaced the CNN initialisation with my model and changed the dataset) and followed the tutorial. 
The only real issue I had using the quantizer (on Vitis-AI 1.3.2) were the unsupported activations (like PReLU and LeakyReLU) after a Conv2d, so I would say that you have to make sure that all your layers are supported (you can find the supported operations on the Vitis-AI manual).

marenan, 2022-08-31:
> 

Thanks for your advices. By the way, if replace LeakyReLU using ReLU, the model need to retrain for quantization?

just-an-intern, 2022-08-31:
> > 
> 
> Thanks for your advices. By the way, if replace LeakyReLU using ReLU, the model need to retrain for quantization?

It depends on the model. For my two models I had drastic losses in accuracy (RetinaFace was unusable after changing the LeakyReLU, the AP dropped to 0 and for another model (for facial verification) I lost 30% of accuracy).

marenan, 2022-08-31:
> 

Thanks!
the Quantization is completed!
But, there are some bugs:

[VAIQ_WARN]: Node ouptut tensor is not quantized: RetinaFace::input_0 type: input
[VAIQ_WARN]: Node ouptut tensor is not quantized: RetinaFace::RetinaFace/IntermediateLayerGetter[body]/Conv2d[conv1]/input.2 type: conv2d
......

just-an-intern, 2022-08-31:
> > 
> 
> Thanks! the Quantization is completed! But, there are some bugs:
> 
> [VAIQ_WARN]: Node ouptut tensor is not quantized: RetinaFace::input_0 type: input [VAIQ_WARN]: Node ouptut tensor is not quantized: RetinaFace::RetinaFace/IntermediateLayerGetter[body]/Conv2d[conv1]/input.2 type: conv2d ......

Could you share the .py output of the quantization script ?
Which version of Vitis-AI are you using ? And which dataset have you used for the calibration ?