Xilinx/Vitis-AI #821: Conv2d output layer DPU limitation: VCK190

AfifaIshtiaq, 2022-05-31:
In the model attached here
[deep_rx_dummy_dilation_rate.zip](https://github.com/Xilinx/Vitis-AI/files/8803813/deep_rx_dummy_dilation_rate.zip)
I can quantize and compile and I get the following output
[image]

The compiler shows 5 subgraphs and 1  DPU graph although I see that all the model is converted to DPU except the output convolution layer. Here is the SVG graph. why the last node is mapped to the CPU?
[image]

qianglin-xlnx, 2022-06-03:
Hi @AfifaIshtiaq 
The output of DPU subgraph is int8. So xcompiler will add a fix2float op in the last node. It's in CPU subgraph and it's expected.
[image]

AfifaIshtiaq, 2022-06-03:
> Hi @AfifaIshtiaq The output of DPU subgraph is int8. So xcompiler will add a fix2float op in the last node. It's in CPU subgraph and it's expected. [image]

Yes you're right so Is it fine to calculate the throughput using the output of DPU graph in the application instead of CPU ? or is there any way to avoid this CPU graph

wanghong4compiler, 2022-06-11:
Hi @AfifaIshtiaq ,
I'm sorry to say the CPU subgraph can not be removed.
The data would be convert from float to INT8 because DPU only supports INT8, but in order to make the output data type of the model is consistent, there should be a CPU subgraph to recover the data type.

AfifaIshtiaq, 2022-06-11:
> Hi @AfifaIshtiaq ,
> I'm sorry to say the CPU subgraph can not be removed.
> The data would be convert from float to INT8 because DPU only supports INT8, but in order to make the output data type of the model is consistent, there should be a CPU subgraph to recover the data type.

So it means it is fine if I set my prediction buffer to dpu graph output instead of cpu output and ignore this output cpu layer?

wanghong4compiler, 2022-07-06:
Hi @AfifaIshtiaq ,
the remove operation is not recommended, but if that's that you need, you could give it a try.

closed 2022-08-11 by qianglin-xlnx